Ultimaker/Cura #1310: Cloud Sync

claell, 2017-01-05:
Since Ultimaker already uses accounts it would be cool to sync profiles etc. to the Cloud so one could use them on different machines.

closed 2018-02-27 by ianpaschal

claell, 2018-02-27:
Could you add an explanation why?

ianpaschal, 2018-02-27:
Sorry, I should have explained.

Because we don't plan on adding this sort of feature for the foreseeable future. Also, if it does, there will be another repository where this sort of thing should be. But it's not going in Cura.

claell, 2018-02-27:
Ok, originally I was looking for having the reason for why you are not planning this explained :smiley: 

Probably it is too much effort to set up a cloud endpoint for this compared to the possible benefits from it?

ianpaschal, 2018-02-27:
Essentially yes. Even now most of our users only have 1 or 2 machines and Cura Connect does a fairly good job at letting a user manage them from one place. I realize that's not the same as managing the settings from multiple computers but we've been working on that in other directions than a cloud service for settings.

Ghostkeeper, 2018-03-21:
If you want a workaround, setting up a symlink to Cura's configuration folder from a network drive or Dropbox or something seems to work all right if all computers have the same operating system. If they don't, you need multiple symlinks and a reasonable knowledge of where Cura's settings are stored.

claell, 2018-03-21:
Yes, that would do the trick. Just wanted to have an easy and convenient way for this.