Ultimaker/Cura #1311: Hollow with most GCode generators ... wrong with CURA 2.3.0

SolidWorksMagi, 2017-01-06:
Hi, 

I'm not sure why the default setting can't build a GCode table better ??? 

Here's the object in the CAD program sliced to show it's hollow;
http://brainless.org/MultiMedia/Pictures/20170105-Rhombicosidodecahedron-Hollow.jpg

Here's the object in the CURA 2.3.0 X-Ray;
http://brainless.org/MultiMedia/Pictures/20170105-Rhombicosidodecahedron-SemiSolid.jpg

And so it's printing those inner squares!
http://brainless.org/MultiMedia/Pictures/IMG_6116-20170105-Rhombicosidodecahedron-Solid-c.jpg

Here's the file ... how do I fix CURA to print it correctly?
http://brainless.org/MultiMedia/Pictures/20mmSolidRhombicosidodecahedron.STL

SolidWorksMagi, 2017-01-06:
Not long after the initial post I looked at the machine again and it's spitting PLA into the air and not printing on the actual part anymore ... somehow the GCode doesn't have a 100% connection on the outer 3mm thick body.  http://brainless.org/MultiMedia/Pictures/IMG_6118-c.jpg

nallath, 2017-01-06:
Your model is completely broken; 
[image]

Ghostkeeper, 2017-01-06:
It's not watertight or has extra faces in the middle. It's ambiguous what parts of the model are on the inside then.

SolidWorksMagi, 2017-01-06:
Hi, 

 

It's solid walls … 

 

3 Views of a Rhombicosidodecahedron

https://www.youtube.com/watch?v=m6HdIxWfTxg 

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7 

 

 <https://cloud.githubusercontent.com/assets/3922611/21712232/c1bcb1f2-d3f3-11e6-813d-cf1b39ed249d.png> solidrhombicosodecahedron

SolidWorksMagi, 2017-01-06:
Hi, 

 

It's solid walls … 

 

3 Views of a Rhombicosidodecahedron

https://www.youtube.com/watch?v=m6HdIxWfTxg 

 

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

thopiekar, 2017-01-06:
@SolidWorksMagi Could you please upload the SolidWorks file?
Next week I'll have access to desktop with SolidWorks installed. I'll try whether I get the same results when converting it into the STL format, if you like.

Thanks!

SolidWorksMagi, 2017-01-06:
Hi, 

 

I posted the link to the .STL file I'm using;

Here's the file ... how do I fix CURA to print it correctly?
 <http://brainless.org/MultiMedia/Pictures/20mmSolidRhombicosidodecahedron.STL> http://brainless.org/MultiMedia/Pictures/20mmSolidRhombicosidodecahedron.STL

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

thopiekar, 2017-01-06:
What about sharing the original file (not the STL - which does not include any useful info or details, except the description of the surface of the model using triangles)? Personally I would be glad if I could understand what is going wrong here.

Thanks!

Ando120, 2017-01-07:
I opened it with Blender and can confirm that this STL is completely broken (mostly missing internal wall parts).

As I already opened it with Blender, then I deleted internal wall and extruded it again .

[fxed20mmSolidRhombicosidodecahedron.zip](https://github.com/Ultimaker/Cura/files/691320/fxed20mmSolidRhombicosidodecahedron.zip)

Opened it with Cura and found new bug. 
Cura 2.4 continuously slicing it about 20 times and then made it.
Progress bar just restart and restart.
Reopened Cura and tested again. Sometimes Cura try 3 or 4 times sometimes much more before its done.

Anyway object is fixed now for SolidWorksMagi

SolidWorksMagi, 2017-01-07:
Hi, 

 

Been busy with CES, but I have the SolidWorks part file uploaded to the 3DContentCentral.com library now … search for SolidWallsRhombicosidodecahedron

 

You can download the file in any version of SolidWorks.

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

SolidWorksMagi, 2017-01-07:
Hi, 

 

Been busy with CES, but I have the SolidWorks part file uploaded to the 3DContentCentral.com library now … search for SolidWallsRhombicosidodecahedron

 

You can download the file in any version of SolidWorks.

 

There are no "inner walls" or any of that other stuff that the CURA 2.3.0 program is generating on its own.  It's a good model and CURA 2.3.0 is the only program that is slicing it up incorrectly.  The model slices just fine in several other 3D printer apps.  

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

thopiekar, 2017-01-07:
This one? http://www.3dcontentcentral.com/search.aspx?arg=solidworksmagi&tagsnavigator=solidworksmagi

SolidWorksMagi, 2017-01-07:
Hi, 

 

That's one of the models that doesn’t print correctly using the CURA 2.3.0 GCode.  

 

I don't know where the other one I uploaded went … I'll upload it again.

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7 

 

This one? http://www.3dcontentcentral.com/search.aspx?arg=solidworksmagi <http://www.3dcontentcentral.com/search.aspx?arg=solidworksmagi&tagsnavigator=solidworksmagi> &tagsnavigator=solidworksmagi

thopiekar, 2017-01-07:
Ok, thanks.

SolidWorksMagi, 2017-01-07:
Hi, 

 

I should have mentioned that that first Rhombicosidodecahedron is the same one except I just closed up the pentagon and square holes and saved it as a different model name.  

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

BagelOrb, 2017-03-08:
The model is broken. Cura handles broken models in different ways than other slicers do, but in many cases we do better.

closed 2017-03-08 by BagelOrb