Ultimaker/Cura #1312: Cura 2.3.1 Starting problems

Reprintable, 2017-01-06:
Hi,

I've been using Cura for over two months without any problems. However, since yesterday my computer keeps crashing when starting the Cura program 2.3.1 @ Updating configuration.... In addition, this problem occurs on both my laptop and desktop. Yesterday I decided to download the BETA version of Cura 2.4, afterwhich I was able to startup the program and setup a new print. Unfortunately, today similair problems occure with this version of Cura. Most of the times my computers crash without giving an error report. Fortunately, this one time I got the following report from Cura on my laptop:

Version: 2.3.1
Platform: Windows-8.1-6.3.9600
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
File "cura_app.py", line 61, in <module>
File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Application.py", line 310, in getInstance
File "D:\2.3\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 173, in __init__
File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 107, in upgrade
File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 281, in _upgradeFile
MemoryError

Can anyone help me out, I'm stuck on how to move on..
Thank you in advance!

Ghostkeeper, 2017-01-06:
It happened in Cura 2.3 when the log file grew too big to even see what type of file it is. To solve, you must clear your log file:

Go to `C:\Users\<Reprintable>\AppData\Local\cura` and delete the file `Cura.log`.

We fixed this in Cura 2.4. It should skip the file if it is too big to load.

closed 2017-01-06 by Ghostkeeper

Reprintable, 2017-01-06:
Thank you for the quick and clear respond. The file was indeed 3.4 GB large.. Cura is running like a pro again!