Ultimaker/Cura #1314: Printing time very wrong on printer (ultimaker2+ firmvare 2.4)

Ando120, 2017-01-06:
"Time left ... hours" is sometimes completely wrong.
Print start with correct time and sometimes its going about 4 or 5 times higher than real total time.
And some point hours dropping like seconds.

Maybe related with "Tune" menu
Last time "Tune"menu was opened about hour while printing and
after return hours was counting up as seconds.

awhiemstra, 2017-01-09:
Printing time is a tricky issue. Cura tries its best to simulate the time it takes to print things. However, it does not (and cannot) account for things like the "print speed" setting on the printer. This may explain the discrepancy. @BagelOrb knows more about the actual details of the time estimation, he spent quite a bit of time getting on them.

BagelOrb, 2017-01-09:
This is most probably a firmware issue. The UM3 actually gets updates in the one estimates each layer. The UM2 works with the total time estimate and uses the amount of bytes of the gcode file processed to get a very rough estimate on the time remaining.

I don't think this issue will ever be fixed for the UM2 family. We're fully engaged with the UM3 now.

Please file this bug under the Ultimaker2_Marlin repo.

closed 2017-01-09 by BagelOrb