Ultimaker/Cura #1315: 2.4 Beta: Wrong Nozzle Temperature Output

CCS86, 2017-01-07:
Bed temp is correctly output, but nozzle temp is posting S255 when it was set to 250.

awhiemstra, 2017-01-09:
A new settings was introduced (Initial bed temperature) that defaults to print temperature + 5. This is probably what you are seeing.

CCS86, 2017-01-13:
@awhiemstra , you are right. There are a bunch of temp settings now!

[image]

I would suggest a way to disable this additional functionality for those who just want one temp, start to finish. If you leave these settings automatically managed, you are opting in for this function. If you set them manually, you need to leave them visible and manually manage the values in them.

Ghostkeeper, 2017-01-22:
We can set `Printing Temperature Initial Layer`, `Initial Printing Temperature` and `Final Printing Temperature` to be equal to `Printing Temperature` by default to clear up confusion for those looking into the g-code or looking closely at the target temperatures on their printer, but then the print quality is slightly worse by default as well. I don't think clearing up the confusion there is worth sacrificing print quality by default.

CCS86, 2017-01-22:
I'm curious what specific improvements in print quality you have seen from these settings?

I can see how increasing for the first layer could improve adhesion and flow, if your chosen temp wasn't high enough. But why is the "initial printing temp" set 10* lower? Why would you want to reduce temp at the end, unless your print tapered at the top?

It still seems odd to jump through these hoops just to set a single print temp.

Ghostkeeper, 2017-01-22:
The `Printing Temperature Initial Layer` was meant to improve bed adhesion, as you guessed.

The `Final Printing Temperature` is meant to slightly cool down the filament just before a nozzle switch. That causes the nozzle to ooze less as it moves away from the model for the nozzle switch.

I'm unsure why `Initial Printing Temperature` was implemented. Do you remember, @BagelOrb?

BagelOrb, 2017-01-23:
Initial temperature was implemented because during heating the material
already absorbs energy. The material stationary at the initial printing
temperature has about the same energy as the filament being pushed through
at the printing temperature.

That's one of the reasons..

fieldOfView, 2017-01-25:
>  I don't think clearing up the confusion there is worth sacrificing print quality by default.

I think it is. Improving print quality is what profiles are for, so this should be part of the print profiles for the UM2+ and UM3. Or it could even be put in the Ultimaker.def.json file to only affect Ultimaker printers.

Ghostkeeper, 2017-01-25:
> Improving print quality is what profiles are for, so this should be part of the print profiles for the UM2+ and UM3.

We also strive for good defaults for other printers. In the end, only a small fraction of users will ever inspect and see that temperature command in the g-code there. More users will have a printer that displays the temperature on its display and might see that it's wrong right when the print is started, but most will only see better bed adhesion and more reliable printing.

fieldOfView, 2017-02-07:
FYI: https://ultimaker.com/en/community/38321-24-beta-2-nozzle-temperature-bug?page=1#unread

> this addition of 5 C to my nozzle temperature could actually have damaged my print head, because I was already running a nozzle temperature that was already right on the line between what the PTFE liner could sustain, while printing PETG, and what it couldn't sustain without cooking the PTFE liner.

If you want the +5 degrees to be a default, do that in the PLA profiles. Doing it blindly for all materials/qualities may break printers, set houses on fire and start WW3.

BagelOrb, 2017-02-10:
> nozzle-temperature-bug?page=1#unread

I've logged a critical bug in JIRA. Let's see if it can still make it
through the pipeline in time before the release.

nallath, 2017-02-10:
The beta was already cleared for release. I don't think this is going to be fixed before then.

tcm0116, 2017-05-12:
> The Final Printing Temperature is meant to slightly cool down the filament just before a nozzle switch. That causes the nozzle to ooze less as it moves away from the model for the nozzle switch.

I'm using an E3D Cyclops hotend, so the `Final Printing Temperature` does nothing but slow down my print. I'm with @CCS86 in that these features should have a way to be disabled. Alternatively, it'd be great if the values could be entered as percentages, an equation, or even entered in the material profile so that I don't have to create different quality profiles for each material type that has different temperatures.

BagelOrb, 2017-05-12:
Altering the print temperature doesn't slow down the print. Note that a set-temperature command is issued, not a set-temperature-and-wait command.

You can simply set the Final PRinting Temperature to the same value as the printing temperature.
IT would indeed be nice if the user could enter formula's, just like we use under the hood.

If you do want formula's you can now enter them in material profiles or in machine definitions.

tcm0116, 2017-05-12:
> Altering the print temperature doesn't slow down the print.

You are correct, except for the very first tool change. I had assumed that was the case for every tool change, but that assumption was incorrect. 

> IT would indeed be nice if the user could enter formula's, just like we use under the hood.

For sure! I like the way OnShape handles variables and equations. When the text for does not have focus, the value is displayed, but when the text field is focused, it shows the equation.

BagelOrb, 2017-05-12:
I think this is the type of thing @fieldOfView can get enthausiastic about...

We might also be able to convince our PO to let us do it if we argue that it makes it much easier for the materials team to work... What do you think, @nallath ?

tcm0116, 2017-05-12:
Don't forget about us end users and how it would make it much easier for us to use the advanced capabilities of Cura currently hidden away under the hood.

awhiemstra, 2017-05-24:
Adding a way to enter formulas in the UI has been on our wishlist for a while actually. It is however not as straightforward as just showing the formula, as the underlying formula actually uses the setting keys which are not exposed to the UI. So we need to come up with an editor widget which allows you to enter setting names and then maps that to setting keys. We also need to do some validation on the formula and a bunch of other things. Since we have more pressing concerns we never really got around to implemeting this.

closed 2018-05-26 by ianpaschal