Ultimaker/Cura #1316: Convert legacy .ini machine profiles to .def.json machine definitions

probonopd, 2017-01-07:
How does one convert legacy `.ini` machine profiles to `.def.json` machine definitions?
I would like to convert [this machine profile](https://gist.githubusercontent.com/probonopd/0dc2afa55fc31caa5ce1f5eea0f3255b/raw/11f440410d935760df20374bcbcbdc47515f03ca/RF100.ini) which came with the machine to the  `.def.json` format that newer versoins of Cura require, __without losing any information__.

I found [LegacyCuraMachineConverter](https://github.com/BagelOrb/LegacyCuraMachineConverter) but it seems unmaintained.

Especially:
* Are there instructions on how to do this, or a converter to do it automatically?
* Are the keys named the same in the `.ini` and `.def.json` formats?
* Are all keys from `.ini` supported in `.def.json` and vice versa?

Using http://quillford.github.io/CuraProfileMaker/ only gives a very rudimentary profile, compared to profiles like [this one](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json).

nallath, 2017-01-09:
There is indeed a difference between a "profile", which describes things like "print this on normal / high quality" or material "pla"/ "abs". The profiles can also describe something like "Abs high quality needs to be printed like this". To even further complicate things, there can be a "Generic ABS high quality" and a "Ultimaker 3 high quality" profile.

There are also machine definitions; These define standards of a machine. Think about physical dimensions, but also settings that due to it's setup should be considered default (eg; Bowden style printers tend to support higher acceleration)

The keys are different. Have a look at the "dictionary of doom". It contains a mapping of all settings from legacy to 2.x. 

As for an automated script that fully uses all of this, i don't think it exists. It's just too complicated to do that. Your best bet is to do it by hand.

awhiemstra, 2017-01-09:
From a brief look at the profile you mentioned, I'd say you want to at least include the [machine] and [alterations] sections of the ini file in the definition. Then you need to figure out what to do with the [profile] section. Some of those can probably be set as new defaults in the definition, but some may simply have become irrelevant.

probonopd, 2017-01-15:
Wrote a converter that translates old `.ini` to new `.def.json` machine definitions:

```
import os, sys, urllib, json, math
import configparser # sudo apt install python-configparser

also_include_default_values = True # Set to True if all values should be converted even if they are default

ini_url = "https://gist.githubusercontent.com/probonopd/0dc2afa55fc31caa5ce1f5eea0f3255b/raw/cb1296e900ff1b05abba5adeb35300d835af2097/RF100.ini"
ini_filename = os.path.basename(ini_url)

dict_url = "https://raw.githubusercontent.com/Ultimaker/Cura/master/plugins/LegacyProfileReader/DictionaryOfDoom.json"
dict_filename = os.path.basename(dict_url)

def get_file(url):
    if not os.path.isfile(os.path.basename(url)):
        urllib.urlretrieve (url, os.path.basename(url))

get_file(ini_url)
get_file(dict_url)

parser = configparser.ConfigParser(interpolation = None)
try:
    with open(ini_filename) as f:
        parser.readfp(f)  # Parse the INI file.
except Exception as e:
    print("Unable to open legacy profile %s: %s", file_name, str(e))

try:
    with open(dict_filename, "r") as f:
        dict_of_doom = json.load(f)  # Parse the Dictionary of Doom.
except IOError as e:
    print("Could not open DictionaryOfDoom.json for reading: %s", str(e))
except Exception as e:
    print("Could not parse DictionaryOfDoom.json: %s", str(e))

if "target_version" not in dict_of_doom:
    print("Dictionary of Doom has no target version. Is it the correct JSON file?")

if "translation" not in dict_of_doom:
    print("Dictionary of Doom has no translation. Is it the correct JSON file?")

def prepareDefaults(json):
    defaults = {}
    for key in json["defaults"]:  # We have to copy over all defaults from the JSON handle to a normal dict.
        defaults[key] = json["defaults"][key]
    return defaults

def prepareLocals(config_parser, config_section, defaults):
    copied_locals = defaults.copy()  # Don't edit the original!
    for option in config_parser.options(config_section):
        copied_locals[option] = config_parser.get(config_section, option)
    return copied_locals

section = ""
for found_section in parser.sections():
    if found_section.startswith("profile"):
        section = found_section
        break
if not section:  # No section starting with "profile" was found. Probably not a proper INI file.
    print("No section starting with 'profile' was found. Probably not a proper INI file.")

defaults = prepareDefaults(dict_of_doom)
legacy_settings = prepareLocals(parser, section, defaults) #Gets the settings from the legacy profile.

data = {}

data['id'] = parser.get('machine', 'machine_name')
data['name'] = parser.get('machine', 'machine_name')
data['version'] = 2 # FIXME: Do not hardcode
[... 408 lines cut ...]
            "default_value": 30.0
        },
        "speed_travel": {
            "default_value": 50.0
        },
        "speed_wall_0": {
            "default_value": 25.0
        },
        "speed_wall_x": {
            "default_value": 35.0
        },
        "support_xy_distance": {
            "default_value": 0.5
        },
        "support_z_distance": {
            "default_value": 0.1
        },
        "top_thickness": {
            "default_value": 0.5
        }
    },
    "version": 2
}
```
I wonder whether I can use the second, shorter version without losing information. Would this machine definition be appropriate for inclusion in Cura?

Ghostkeeper, 2017-01-22:
Our legacy profile importer (which is not intended for machine profiles) will use Cura 2's default when a setting is equal to Legacy Cura's default too. It'll get the speed "wrong" when compared to the speed it produced in Legacy Cura, but the idea is that we find better defaults all the time and unless the user specifically meant to change a setting, the new defaults should work better than the old ones. Often the new defaults tie in with new features of the slicer or changed functionality.

probonopd, 2017-01-22:
@Ghostkeeper thanks for your response, I am curious to learn more. How would you translate the speed settings from [this `.ini`](https://gist.githubusercontent.com/probonopd/0dc2afa55fc31caa5ce1f5eea0f3255b/raw/11f440410d935760df20374bcbcbdc47515f03ca/RF100.ini)?

Can you spot mistakes in my attempted translation [here](https://github.com/probonopd/Cura/blob/patch-1/resources/definitions/renkforce_rf100.def.json)?

Ghostkeeper, 2017-01-22:
With `also_include_default_values` to `False` the result looks all right. There is one crucial mistake though, which is easily fixed in your script: You should overwrite `value` rather than `default_value`.

`value` overrides `default_value` if it is set. For some settings `value` is set and for some settings it isn't. So for some settings you change the `default_value` and you'll see the result in Cura, but for other settings you'll change the `default_value` but it still gets overwritten by the old `value`. It is safer to always overwrite `value`.

It's difficult to check all those settings. I advise you to go by what the LegacyProfileReader plug-in does. It is basically the same as what you do but behaves differently when modern Cura 2's defaults change, as I explained.

probonopd, 2017-01-22:
How is the LegacyProfileReader intended to be used? Is converting old machine profiles something you are working on as a Cura feature?

probonopd, 2017-01-22:
> You should overwrite value rather than default_value (and make the result in the JSON a string then).

@Ghostkeeper looking at the other definition files in https://github.com/Ultimaker/Cura/blob/master/resources/definitions I don't quite understand this comment -- the other definition files all seem to use `default_value`? For example [here](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/prusa_i3_mk2.def.json#L17-L52). Should I really replace all occurences of `default_value` with `value`?

Ghostkeeper, 2017-01-23:
Indeed, those are wrong too. We usually aren't able to properly test third-party printers since we have no physical printer to test with. For machine settings it is quite safe to just use `default_value` since we don't use inheritance (yet). But for others it's not, and things did break since the printer definitions were added. For example, I'd expect the Travel Speed of the PrintrBot Play to not work: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/printrbot_play.def.json#L32

You could look for when a `value` is defined or when a `default_value` is allowed but it's easier and safer for the future to always use `value`. We did that for the Ultimaker 3 for that reason too (that file was also automatically generated).

When you replace `default_value` with `value`, be sure to convert the content of that setting to a string that evaluates to the same value in Python. It is evaluated with an `eval()` call. So `3.14` becomes `"3.14"` and `true` becomes `"True"`.

probonopd, 2017-01-23:
I hope I did that right in https://github.com/probonopd/Cura/commit/e7cad12bf612fbf4c8c78d87cab8ab896776ca9f

If there is interest, the Python script above could be changed to reflect this, too.

BagelOrb, 2017-03-08:
I made a crappy ass script over here: https://github.com/BagelOrb/LegacyCuraMachineConverter
No guarantees and no manual, though :/
Still it might be of some use..

closed 2017-07-27 by nallath