Ultimaker/Cura #1318: python3.4: command not found, and ImportError: cannot import name 'multiarray'

Merlin04, 2017-01-08:
```
username@my-linux-box (bash) ~ $ cura
/usr/bin/cura: line 3: python3.4: command not found
```
/usr/bin/cura reads:
```
#!/bin/bash

LD_LIBRARY_PATH=/opt/cura/lib PYTHONPATH=/opt/cura/lib/python3/dist-packages QT_PLUGIN_PATH=/opt/cura/lib/plugins QML2_IMPORT_PATH=/opt/cura/lib/qml QT_QPA_FONTDIR=/opt/cura/lib/fonts python3.4 /opt/cura/bin/cura "$@"
```

When I change `python3.4` to `python3.5`, I get this error:
```
username@my-linux-box (bash) ~ $ cura
Traceback (most recent call last):
  File "/opt/cura/bin/cura", line 10, in <module>
    from UM.Platform import Platform
  File "/opt/cura/lib/python3/dist-packages/UM/__init__.py", line 9, in <module>
    from .Application import Application
  File "/opt/cura/lib/python3/dist-packages/UM/Application.py", line 9, in <module>
    from UM.Controller import Controller
  File "/opt/cura/lib/python3/dist-packages/UM/Controller.py", line 4, in <module>
    from UM.Scene.Scene import Scene
  File "/opt/cura/lib/python3/dist-packages/UM/Scene/Scene.py", line 4, in <module>
    from UM.Scene.SceneNode import SceneNode
  File "/opt/cura/lib/python3/dist-packages/UM/Scene/SceneNode.py", line 4, in <module>
    from UM.Math.Matrix import Matrix
  File "/opt/cura/lib/python3/dist-packages/UM/Math/Matrix.py", line 7, in <module>
    import numpy
  File "/opt/cura/lib/python3/dist-packages/numpy/__init__.py", line 170, in <module>
    from . import add_newdocs
  File "/opt/cura/lib/python3/dist-packages/numpy/add_newdocs.py", line 13, in <module>
    from numpy.lib import add_newdoc
  File "/opt/cura/lib/python3/dist-packages/numpy/lib/__init__.py", line 8, in <module>
    from .type_check import *
  File "/opt/cura/lib/python3/dist-packages/numpy/lib/type_check.py", line 11, in <module>
    import numpy.core.numeric as _nx
  File "/opt/cura/lib/python3/dist-packages/numpy/core/__init__.py", line 6, in <module>
    from . import multiarray
ImportError: cannot import name 'multiarray'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 63, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
  File "/usr/lib/python3/dist-packages/apport/__init__.py", line 5, in <module>
    from apport.report import Report
  File "/usr/lib/python3/dist-packages/apport/report.py", line 30, in <module>
    import apport.fileutils
  File "/usr/lib/python3/dist-packages/apport/fileutils.py", line 23, in <module>
    from apport.packaging_impl import impl as packaging
  File "/usr/lib/python3/dist-packages/apport/packaging_impl.py", line 23, in <module>
    import apt
  File "/usr/lib/python3/dist-packages/apt/__init__.py", line 23, in <module>
    import apt_pkg
ImportError: /opt/cura/lib/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/lib/python3/dist-packages/apt_pkg.cpython-35m-x86_64-linux-gnu.so)

Original exception was:
Traceback (most recent call last):
  File "/opt/cura/bin/cura", line 10, in <module>
    from UM.Platform import Platform
  File "/opt/cura/lib/python3/dist-packages/UM/__init__.py", line 9, in <module>
    from .Application import Application
  File "/opt/cura/lib/python3/dist-packages/UM/Application.py", line 9, in <module>
    from UM.Controller import Controller
  File "/opt/cura/lib/python3/dist-packages/UM/Controller.py", line 4, in <module>
    from UM.Scene.Scene import Scene
  File "/opt/cura/lib/python3/dist-packages/UM/Scene/Scene.py", line 4, in <module>
    from UM.Scene.SceneNode import SceneNode
  File "/opt/cura/lib/python3/dist-packages/UM/Scene/SceneNode.py", line 4, in <module>
    from UM.Math.Matrix import Matrix
  File "/opt/cura/lib/python3/dist-packages/UM/Math/Matrix.py", line 7, in <module>
    import numpy
  File "/opt/cura/lib/python3/dist-packages/numpy/__init__.py", line 170, in <module>
    from . import add_newdocs
  File "/opt/cura/lib/python3/dist-packages/numpy/add_newdocs.py", line 13, in <module>
    from numpy.lib import add_newdoc
  File "/opt/cura/lib/python3/dist-packages/numpy/lib/__init__.py", line 8, in <module>
    from .type_check import *
  File "/opt/cura/lib/python3/dist-packages/numpy/lib/type_check.py", line 11, in <module>
    import numpy.core.numeric as _nx
  File "/opt/cura/lib/python3/dist-packages/numpy/core/__init__.py", line 6, in <module>
    from . import multiarray
ImportError: cannot import name 'multiarray'
```

awhiemstra, 2017-01-09:
The .deb package unfortunately does not support Ubuntu 16.04 or higher. There are a few options though: 

- Use the AppImage build, which will be the only supported version from 2.4 forward. http://software.ultimaker.com/current/Cura-2.3.1.AppImage for the 2.3.1 version. The 2.4 Beta version is available from the Ultimaker site directly: https://ultimaker.com/en/products/cura-software/list . 
- Use @thopiekar 's PPA. See https://github.com/Ultimaker/Cura/issues/718 for some details on that.

closed 2017-01-09 by awhiemstra

thopiekar, 2017-01-10:
Correct, also https://github.com/thopiekar/Cura-packaging/blob/master/README.md is a good start into the PPA installation.

Good luck @Merlin04