Ultimaker/Cura #1321: 2.4 beta USB issues

tjcolejr, 2017-01-09:
I have 2 printers (one is a printer/cnc) connected via USB to my computer.  Cura does not identify them specifically by either name or com port.  There are just two lines in the drop down menu that say "Print to USB."  I figured the first one listed would be the lower com port number, but I was wrong.
Also, it would be wonderful if Cura did not automatically grab every connected printer upon startup.  If I am printing from an SD card and start Cura on my pc, which is connected via USB, the SD card print gets killed.   I can work around this by leaving Octoprint and Repetier connected to the printer and cnc respectively or just unplugging the USB cables before running Cura, but I'd rather have a checkbox in Cura to connect to USB on startup, so I can decide when I want Cura connected.

fieldOfView, 2017-01-09:
See https://github.com/Ultimaker/Cura/issues/1049

leutholl, 2017-04-05:
In USBPrinterOutputDeviceManager.py I changed the tty filter to be /dev/tty.* in order to also accept kits that list their tty als tty.* under /dev

change line 274 so that the else: statement looks like this
        else:
            if only_list_usb:
                base_list = base_list + glob.glob("/dev/tty.*") + glob.glob("/dev/ttyACM*") + glob.glob("/dev/cu.usb*")
                base_list = filter(lambda s: "Bluetooth" not in s, base_list) # Filter because mac sometimes puts them in the list
            else:
                base_list = base_list + glob.glob("/dev/ttyUSB*") + glob.glob("/dev/ttyACM*") + glob.glob("/dev/cu.*") + glob.glob("/dev/tty.usb*") + glob.glob("/dev/rfcomm*") + glob.glob("/dev/serial/by-id/*")
        return list(base_list)

    _instance = None

nallath, 2017-04-05:
If you have code changes, please make a pull request for it. That way others can also benefit from your changes.

leutholl, 2017-04-05:
Hi.

Yeah sorry. I'm sorry to say I don't know how all this pull request work... maybe somebody can pick my idea and make it in the next release?!

tjcolejr, 2017-04-05:
leutholl, thank you for your input.

ianpaschal, 2018-02-21:
No more work will be done on versions 2.x. If this is still occurring in versions 3.2 or newer, we can re-open this issue.

closed 2018-02-21 by ianpaschal

Ghostkeeper, 2018-03-13:
For release 2.6 we made the serial port detection use Python's built-in serial module which solved the issue that leutholl's proposed code change would solve as well. The rest should be handled in #1049.