Ultimaker/Cura #1322: After start printing, heated bed does not want to heat up (UM3, CURA 2.4.0)

landway, 2017-01-09:
Hey,

I have strange problem on start of print batch. When I execute print from USB, heated bed does not heat up. In tune menu I see 22/80C, but temperature is not change during time. After cca 10 minutes I stop print.

PRINT BATCH:
[UM3_ABSSestava s kl-X-E1vesnic-X-ED.gcode.txt](https://github.com/Ultimaker/Cura/files/693789/UM3_ABSSestava.s.kl-X-E1vesnic-X-ED.gcode.txt)
CURA LOGs:
[UM3 CURA LOG.zip.txt](https://github.com/Ultimaker/Cura/files/693797/UM3.CURA.LOG.zip.txt)
LOGs from UM3:
[UM3 SHOWROOM.zip.txt](https://github.com/Ultimaker/Cura/files/693791/UM3.SHOWROOM.zip.txt)

landway, 2017-01-09:
Same problem when sending print batch over wifi...
[image]

awhiemstra, 2017-01-09:
From the G-Code:

`;BUILD_PLATE.INITIAL_TEMPERATURE:80`

So Cura is sending the right data, this suggests either an embedded or hardware issue. The fact that the printer's menu indicates 80C strongly suggests a hardware problem. I would recommend contacting support and having the hardware checked. In any case, I do not think it is something we can do a lot about in Cura.

landway, 2017-01-10:
I tried factory reset and now it is printing. (while rebooting printer it stucks - i must manualy restart)

Does anyone know where is github for FW UM3?

awhiemstra, 2017-01-10:
There is no public one yet, as far as I know. @daid might know something more about this. (I'm closing this issue since the problem was resolved).

closed 2017-01-10 by awhiemstra

daid, 2017-01-10:
It's a known problem right now, sometimes (rarely) during starting of a print, a temperature set command is not handled properly. A workaround is tuning a 1C offset, and then tune it back to 0C, at which point the command is handled properly.

landway, 2017-01-10:
Thank you guys.