Ultimaker/Cura #1325: Cura 2.3.1 Stacks copies of model in same place

ianlpowell, 2017-01-10:
When I load the attached stl file multiple times or multiply it Cura puts them in the same position. You can move them separately so they have loaded.

[Poster_Coner_3.zip](https://github.com/Ultimaker/Cura/files/695872/Poster_Coner_3.zip)

Running in windows 10  ver 1607

BerndJM, 2017-01-10:
Have a look at:
Preferences -> "Ensure models are kept apart"

Ghostkeeper, 2017-01-22:
Did activating the "Ensure models are kept apart" setting fix your problem, @ianlpowell?

ianlpowell, 2017-01-24:
Hi, sorry for the delay in coming back to you.
The "Ensure models are kept apart" was and is ticked!
For all my other models I have printed stacking is not a problem, it only happens with the model I attached to this post.

Ghostkeeper, 2017-01-24:
Ah, I could reproduce this. For some reason, it doesn't seem to produce the convex hull collision area for this mesh. Though QHull gives no errors. That's a bug.

ChrisTerBeke, 2017-11-14:
Is this still an issue?

Ghostkeeper, 2017-11-23:
No, the model arranger seems to have taken care of this bug.

closed 2017-11-23 by Ghostkeeper