Ultimaker/Cura #1327: layer_count and layer_number as variables

maukcc, 2017-01-10:
Is there a way to get layer_count and layer_number as variables?
Then you could output that to the printer screen like:
M117 Printing layer {layer_number} / {layer_count}

Ghostkeeper, 2017-01-22:
No, that is not possible, or at least not trivially. These variables are replaced with their values upon sending the mesh to the slicer, so they need to be known before slicing. The number of layers (and the layer count) is only known after slicing.

It requires quite intensive computation to find the number of layers, because we need to find out what layers are empty; CuraEngine moves the mesh down if the bottom-most layers are empty. Whether a layer is empty is only known after drawing the actual lines since a mesh may be too slim to fit any wall lines in.

So the only real solution is to let the engine fill these variables in during slicing instead of letting the front-end do that beforehand. And that is quite a refactor.

maukcc, 2017-01-23:
OK, no problem. I was just thinking of ways to let the user have a sense of how long the print would need.
Please close this issue.

closed 2017-01-23 by Ghostkeeper