Ultimaker/Cura #1328: 2.4b: duplicating a Material will not create a new GUID

Spirou42, 2017-01-10:
Creating a new material preset  by pressing the [Duplicate] button creates a new material which contain the same /fdmmaterial/metadata/GUID entry in the DOM. If GUID stands for  "Global Unique Identifier" this is bug and might be the cause for bugs like #1286  or #1238.

nallath, 2017-01-10:
You are duplicating a profile. Not duplicating a material. A GUID identifies a material. In this case, both profiles talk about the same material, so should share the same GUID. 

I'm not a big fan of how the materials work right now and I expect them to get some love in the next update. Adding a new material (aka; something that also spawns a new GUID) is something that we want to build at some point in the future.

Spirou42, 2017-01-11:
So the GUID tag in a FDM-Material-XML is not a GUID but a reference to another FDM-Material entity. In my case a FDM-Material described inside the Cura.app bundle which has absolutely the same structure.
How do you determine which of the Descriptions is the source and which one is the destination? By its position in the file system?

If i use one the manually created FDM-Material in an also newly created Profile (I call the entity created in the [Profile] section of the UI a profile for now) ALL Profiles disappear switching back to a default material (you have to pick a material with a GUID wich is really unique, not the one you have base the custom Material creation upon) magically the profiles appear again.
Changing the GUID in the custom Material FDM-Material file in such a way, that the GUID is really unique, every thing is ok.

Ghostkeeper, 2017-01-22:
The GUID is a reference to an actual physical material. Not to some other entity on the file system. Cura uses it for the Ultimaker 3 to sync via wifi which materials are loaded on the printer. So in that case, it essentially references a material in the RFID tag of the spool of the material.

We used to require that all GUIDs are also unique in Cura, until we implemented duplicated materials and found out that this is a problem. We might have some bugs leftover from this switch, such as #1238 that you mentioned.

Spirou42, 2017-01-22:
So this references actual RFID IDs. Than there has to be a way to create GUID free materials because it would be utterly incorrect to create an entry for nGen or HT by duplicating any Ultimaker-Material cause this will copy a an incorrect GUID into the Material-File.

Ghostkeeper, 2017-01-22:
I'd like to allow editing the GUID manually from the interface, indeed. Perhaps with a button to randomise the GUID.

peteruithoven, 2017-02-01:
My issue seems related to this one as well: https://github.com/Ultimaker/Cura/issues/1352

Spirou42, 2017-02-01:
The absolut optimum would be an export of the custom material as a NDEF dump so any body could create there own Mifare-Ultralight tag and an upload possibility of the necessary data to the Ultimaker.

Ghostkeeper, 2017-06-05:
Our product owner removed this issue from our planning for being inactive for 3 months.

awhiemstra, 2017-06-06:
Additionally, we now have a way of actually creating "new" materials that will assign a new GUID to the material.

nallath, 2017-06-06:
You can also unlink duplicated materials. So it's actually something that's fixed.

closed 2017-06-06 by nallath