Ultimaker/Cura #1329: 2.4.0 Beta for Linux: Sending anonymous data FAILs

Jarek-SRT, 2017-01-10:
Cura 2.4.0 Beta packaged as AppImage for LInux cannot send anonymous slice info
`2017-01-10 21:54:00,590 - DEBUG - SliceInfoPlugin.SliceInfo.run [50]: Sending anonymous slice info to [https://stats.youmagine.com/curastats/slice]...
2017-01-10 21:54:00,732 - ERROR - SliceInfoPlugin.SliceInfo.run [59]: An exception occurred while trying to send slice information: <urlopen error A failure in the SSL library occurred (_ssl.c:645)>`

awhiemstra, 2017-01-11:
Hmm, that's a tricky one, it seems the SSL connection fails for some reason. What distribution are you running on?

Jarek-SRT, 2017-01-11:
I'm running openSUSE Leap 42.2. It has Internet acces and I have no problem making outgoing connections from others apps. If you suspect the outgoing connection is blocked, I will look into my system to see if anything on my side might be blocking this particular connection.

awhiemstra, 2017-01-11:
No, I suspect there is something broken with the SSL library used to make the connection. Though you could verify that you can connect to https://stats.youmagine.com .

Jarek-SRT, 2017-01-11:
I can connect via SSL to stats.youmagine.com from my system (but cura cannot). Full dump from curl below, but basically it shows that the SSL connection is properly established and the server responds with 401 requesting authentication, which I assume is the the expected behaviour.

```
> curl -v https://stats.youmagine.com    
* Rebuilt URL to: https://stats.youmagine.com/
* Hostname was NOT found in DNS cache
*   Trying 188.166.83.224...
* Connected to stats.youmagine.com (188.166.83.224) port 443 (#0)
* successfully set certificate verify locations:
*   CAfile: none
  CApath: /etc/ssl/certs/
* SSLv3, TLS Unknown, Unknown (22):
* SSLv3, TLS handshake, Client hello (1):
* SSLv2, Unknown (22):
* SSLv3, TLS handshake, Server hello (2):
* SSLv2, Unknown (22):
* SSLv3, TLS handshake, CERT (11):
* SSLv2, Unknown (22):
* SSLv3, TLS handshake, Server key exchange (12):
* SSLv2, Unknown (22):
* SSLv3, TLS handshake, Server finished (14):
* SSLv2, Unknown (22):
* SSLv3, TLS handshake, Client key exchange (16):
* SSLv2, Unknown (20):
* SSLv3, TLS change cipher, Client hello (1):
* SSLv2, Unknown (22):
* SSLv3, TLS handshake, Finished (20):
* SSLv2, Unknown (20):
* SSLv3, TLS change cipher, Client hello (1):
* SSLv2, Unknown (22):
* SSLv3, TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / ECDHE-RSA-AES128-GCM-SHA256
* Server certificate:
*        subject: OU=GT05874481; OU=See www.rapidssl.com/resources/cps (c)14; OU=Domain Control Validated - RapidSSL(R); CN=*.youmagine.com
*        start date: 2014-10-21 15:09:56 GMT
*        expire date: 2017-08-11 12:30:54 GMT
*        subjectAltName: stats.youmagine.com matched
*        issuer: C=US; O=GeoTrust Inc.; CN=RapidSSL SHA256 CA - G3
*        SSL certificate verify ok.
* SSLv2, Unknown (23):
 > GET / HTTP/1.1 
 > User-Agent: curl/7.37.0
 > Host: stats.youmagine.com
 > Accept: */*
 > 
* SSLv2, Unknown (23):
< HTTP/1.1 401 Unauthorized
< X-Frame-Options: SAMEORIGIN
< X-XSS-Protection: 1; mode=block
< X-Content-Type-Options: nosniff
< WWW-Authenticate: Basic realm="Application"
< Content-Type: text/html; charset=utf-8
< Cache-Control: no-cache
< X-Request-Id: 43ee0e32-633d-48d6-8e08-bb2c6a1d563b
< X-Runtime: 0.004201
< Transfer-Encoding: chunked
```
<

Jarek-SRT, 2017-01-11:
OK, it seems that the issue disappeared. Cura can now connect to the server. I might have been a server issue. Closing and apologies for rising a false alarm.

closed 2017-01-11 by Jarek-SRT

awhiemstra, 2017-01-11:
That's fine, the server is occassionally wonky. Thanks for reporting!