Ultimaker/Cura #1330: 2.4.0 Beta for Linux: segmentation fault in engine on Cura exit

Jarek-SRT, 2017-01-10:
This happens in Cura 2.4.0 Beta packaged as AppImage for Linux. Exiting Cura by choosing from the menu File -> Quit results in Segmentation fault. This happens every time, steps to reproduce are just: 

1. Start the app from AppImage from terminal (so you can see the error)
2. Choose File -> Quit 
3. Look into terminal and see the Segmentation fault:
`2017-01-10 22:13:52,442 - DEBUG - UM.Qt.QtApplication.windowClosed [198]: Shutting down cura
2017-01-10 22:13:54,861 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [197]: Attempting to kill the engine process
2017-01-10 22:13:54,861 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [203]: Killing engine process
2017-01-10 22:13:54,874 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [206]: Engine process is killed. Received return code -15
./bin//cura.sh: line 11:  1442 Segmentation fault      (core dumped) cura $@`

awhiemstra, 2017-01-11:
I have seen it too. It is not a crash in the engine but in Cura itself, probably related to Qt and Python fighting over who destroys what. Since it only ever happens on application shutdown, and after everything else is already done, it does not have a very high priority for me. It is also not the simplest thing to reproduce.

awhiemstra, 2017-01-11:
However, thank you for reporting it. :)

Jarek-SRT, 2017-01-11:
No problem, I fully agree that's not a high priority issue, as the only way it affects end-users is showing the message.

Ghostkeeper, 2017-01-22:
This one has happened for a while. Like, a year, or something.

hroncok, 2017-05-03:
Valgrind says:

```
==17124== Thread 5 QDBusConnectionManager:
==17124== Invalid read of size 8
==17124==    at 0x15036F8D: QObject::disconnect(QObject const*, char const*, QObject const*, char const*) (in /usr/lib64/libQt5Core.so.5.7.1)
==17124==    by 0x27EDEFAF: ??? (in /usr/lib64/libQt5DBus.so.5.7.1)
==17124==    by 0x27ECB851: ??? (in /usr/lib64/libQt5DBus.so.5.7.1)
==17124==    by 0x14E6C9C9: ??? (in /usr/lib64/libQt5Core.so.5.7.1)
==17124==    by 0x53106C9: start_thread (in /usr/lib64/libpthread-2.24.so)
==17124==  Address 0x37eafaa0 is not stack'd, malloc'd or (recently) free'd
==17124== 
==17124== 
==17124== Process terminating with default action of signal 11 (SIGSEGV)
==17124==  Access not within mapped region at address 0x37EAFAA0
==17124==    at 0x15036F8D: QObject::disconnect(QObject const*, char const*, QObject const*, char const*) (in /usr/lib64/libQt5Core.so.5.7.1)
==17124==    by 0x27EDEFAF: ??? (in /usr/lib64/libQt5DBus.so.5.7.1)
==17124==    by 0x27ECB851: ??? (in /usr/lib64/libQt5DBus.so.5.7.1)
==17124==    by 0x14E6C9C9: ??? (in /usr/lib64/libQt5Core.so.5.7.1)
==17124==    by 0x53106C9: start_thread (in /usr/lib64/libpthread-2.24.so)
==17124==  If you believe this happened as a result of a stack
==17124==  overflow in your program's main thread (unlikely but
==17124==  possible), you can try to increase the size of the
==17124==  main thread stack using the --main-stacksize= flag.
==17124==  The main thread stack size used in this run was 8388608.
==17124== 
==17124== HEAP SUMMARY:
==17124==     in use at exit: 222,693,258 bytes in 1,798,048 blocks
==17124==   total heap usage: 13,769,268 allocs, 11,971,220 frees, 4,861,666,942 bytes allocated
==17124== 
==17124== LEAK SUMMARY:
==17124==    definitely lost: 4,765,808 bytes in 3,176 blocks
==17124==    indirectly lost: 145,890 bytes in 1,454 blocks
==17124==      possibly lost: 25,470,256 bytes in 233,199 blocks
==17124==    still reachable: 192,246,304 bytes in 1,559,689 blocks
==17124==                       of which reachable via heuristic:
==17124==                         length64           : 3,376 bytes in 61 blocks
==17124==                         newarray           : 817,455 bytes in 2,435 blocks
==17124==                         multipleinheritance: 148,952 bytes in 615 blocks
==17124==         suppressed: 0 bytes in 0 blocks
==17124== Rerun with --leak-check=full to see details of leaked memory
==17124== 
==17124== For counts of detected and suppressed errors, rerun with: -v
==17124== Use --track-origins=yes to see where uninitialised values come from
==17124== ERROR SUMMARY: 384 errors from 3 contexts (suppressed: 0 from 0)
Segmentation fault
```

hroncok, 2017-05-03:
Might be related to https://bugreports.qt.io/browse/QTBUG-52988

hroncok, 2017-05-30:
[This automatically reported bug from Fedora](https://bugzilla.redhat.com/show_bug.cgi?id=1456733) contains the [full backtrace](https://bugzilla.redhat.com/attachment.cgi?id=1283369).

ianpaschal, 2018-02-21:
No more work will be done on versions 2.x. If this is still occurring in versions 3.2 or newer, we can re-open this issue.

closed 2018-02-21 by ianpaschal

Ghostkeeper, 2018-03-13:
We haven't seen this bug any more since we upgraded to Qt 5.8, so hroncok was probably right.