Ultimaker/Cura #1331: Mouse clicks incompletely captured leaving controls in mouse down state

samdale12, 2017-01-11:
When manipulating the model in the viewer (rotating, moving, scaling or rotating view) it seems as though the mouse click (button down) is captured, but the mouse release is either not captured or is lost, leaving the tool active, and resulting in an inability to move the mouse within the window without changing the parameters (either rotating the object, moving the object, scaling the object or rotating the view, depending on which tool was selected). Clicking the mouse button again does not resolve the issue - the only effective solution is to CMD-Q quit the application and reopen.
Happens with a relatively high frequency in Cura for Mac 2.3.1.

nallath, 2017-01-11:
Does it also happen in the beta 2.4.0 version?

samdale12, 2017-01-11:
I have to confess I hadn't tried 2.4.0, but I have just tried it - I wasn't able to get trapped in any of the transform modes, but did get 'mouse-trapped' in the rotate-view mode (ie; anytime I moved my mouse the view would rotate, new clicks, space-bar etc, could not get me out of this mode). I tried to get the behaviour to happen in each of the transform modes, but it didn't happen.
I'll keep trying it and report back.

Ghostkeeper, 2017-01-23:
I can't reproduce this. I tried releasing the mouse on top of various interface elements while dragging the object in rotation. It always seems to properly release the rotation.

Tested on Ubuntu 16.10 using the current state of the 2.4 branch.

ianpaschal, 2018-02-27:
This is no longer an issue in 3.2.1

closed 2018-02-27 by ianpaschal