Ultimaker/Cura #1332: Print hung while printing from Cura after executing start gcode preamble.

jvillar, 2017-01-11:
My printer use Ramps 1.4 with Marlin. I have a capacitive proximity detector and z-safe homing enabled on Marlin. When It hangs, the last line of my LCD says: "05 TO"

If I export the Gcode from Cura and import it on RepetierHost, the print doesn't hang and everything finish OK. 

My start Gcode is the following:
"""
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops, so the head is out of the way
G1 F200 E3 ;extrude 3mm of feed stock
G1 F200 E-2   ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G92 E0 ;zero the extruded length again

G28 ;move XYZ to min endstops
G29 T S4
G1 X20 Y45 F9000; Move to the initial position. X0,Y0 on my heatbed.
G92 X0 Y0 ; Z4.35 Set this coordinate to X0,Y0

G1 Z15.0 F9000 ;move the platform down 15mm
G1 F9000
M117 Printing...
"""
My end Gcode is the following:
"""
M104 S0 ;extruder heater off
M140 S0 ;heated bed heater off (if you have it)
G91 ;relative positioning
G1 E-1 F300  ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more
G28 X0 Y0 ;move X/Y to min endstops, so the head is out of the way
M84 ;steppers off
G90 ;absolute positioning
"""

awhiemstra, 2017-01-11:
I am afraid I do not really have an idea of what is going wrong here. Do you print with USB? Because in that case your machine might not like some of the things we do with the USB connection.

jvillar, 2017-01-13:
Hi again!
I tested 2.4 beta and everything works again. Don't know what happened but it's been fixed in the next release. 

Thanks!

awhiemstra, 2017-01-16:
In that case, I will close this issue.

closed 2017-01-16 by awhiemstra