Ultimaker/Cura #1333: Replacing filament at middle of printing

kastielt, 2017-01-11:
Hi, I tried to pause the printing and replace filament to nother PLA color.
Everything went OK at the begining. The old filament went outside but when tried to enter the new one, the system wrote "Heating/Cooling" and did not arrive the stage new filament starts geting inside

awhiemstra, 2017-01-11:
Are you using an Ultimaker 2? If so, and the pause happened correctly, there is unfortunately not much more we can do from the Cura side. The material change procedure is part of the firmware of the machine which we have no control over. You can have a look at https://github.com/Ultimaker/Ultimaker2Marlin which is the repository for the UM2 firmware.

kastielt, 2017-01-11:
Thanks for your reply. This is a new Ultimaker 2+. bought it 2 weeks ago. The screen changes from Heating to cooling and back rapidy as if it doesn't reach the temperature. I tried playing with the temperature but it didn't help. Do you think it's a problem with the machine ?

kastielt, 2017-01-11:
When I change a filament not in middle of printing, everything is just fine

awhiemstra, 2017-01-11:
It sounds to me like a bug in the firmware. Maybe @daid has an idea what's going wrong?

daid, 2017-01-11:
Yes, the heating/cooling is a bug in the UM2 firmware when trying to change material during a paused print. Cura 2.4 will ship with an UM2 firmware with a fix for this bug.

kastielt, 2017-01-11:
OK. Thank you. Have an estimation of time when will Cura 2.4 be shipped ?

awhiemstra, 2017-01-11:
It is available as beta at the moment. The expected final release is at the end of the month.

kastielt, 2017-01-11:
Thank you very much !

kastielt, 2017-01-12:
Tried the Cura 2.4 . Problem solved. However, when extruder starts workung with 2nd filament (after I waited filament remains go out of the extruder when filament was replaced) it flows out some filamet before starting to work on plane.

awhiemstra, 2017-01-12:
That is controlled by the "Extrude Amount" option in the post-processing script, you will need to tweak that value to get the right amount. In any case, since this issue has been resolved with the new firmware, I am closing it.

closed 2017-01-12 by awhiemstra