Ultimaker/Cura #1334: cura 2.4 segfaults during startup, ubuntu 16.10

grg2rsr, 2017-01-11:
As the title says, cura 2.4 freshly installed from the ppa on a ubuntu 16.10, doesn't start, segfault before the UI shows up.
Here is the console output. Any help is appreciated.
```shell
(root) georg@georg-desktop:~$ cura
2017-01-11 14:27:02,987 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ConsoleLogger
2017-01-11 14:27:03,001 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /usr/bin/CuraEngine
2017-01-11 14:27:03,001 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CuraEngineBackend
2017-01-11 14:27:03,004 - WARNING - STLReader.STLReader.<module> [26]: Could not find numpy-stl, falling back to slower code.
2017-01-11 14:27:03,005 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin STLReader
2017-01-11 14:27:03,007 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin OBJWriter
2017-01-11 14:27:03,008 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin STLWriter
2017-01-11 14:27:03,011 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin OBJReader
2017-01-11 14:27:03,013 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LocalFileOutputDevice
2017-01-11 14:27:03,014 - WARNING - UM.PluginRegistry.loadPlugin [72]: Plugin ConsoleLogger was already loaded
2017-01-11 14:27:03,018 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ScaleTool
2017-01-11 14:27:03,025 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin RotateTool
2017-01-11 14:27:03,028 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin TranslateTool
2017-01-11 14:27:03,030 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CameraTool
2017-01-11 14:27:03,033 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin MirrorTool
2017-01-11 14:27:03,035 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SelectionTool
2017-01-11 14:27:03,037 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SimpleView
2017-01-11 14:27:03,039 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin WireframeView
2017-01-11 14:27:03,041 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin FileLogger
2017-01-11 14:27:03,042 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin GCodeWriter
2017-01-11 14:27:03,045 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ChangeLogPlugin
2017-01-11 14:27:03,047 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin XRayView
2017-01-11 14:27:03,106 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin USBPrinting
2017-01-11 14:27:03,113 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CuraProfileWriter
2017-01-11 14:27:03,137 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin X3DReader
2017-01-11 14:27:03,139 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LegacyProfileReader
2017-01-11 14:27:03,143 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin VersionUpgrade21to22
2017-01-11 14:27:03,145 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin VersionUpgrade22to24
2017-01-11 14:27:03,153 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UltimakerMachineActions
2017-01-11 14:27:03,182 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SliceInfoPlugin
2017-01-11 14:27:03,184 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin GCodeProfileReader
2017-01-11 14:27:03,187 - WARNING - UM.Preferences.getValue [81]: Tried to get the value of non-existing setting cura/active_mode.
2017-01-11 14:27:03,187 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin PerObjectSettingsTool
2017-01-11 14:27:03,190 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SolidView
2017-01-11 14:27:03,275 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UM3NetworkPrinting
2017-01-11 14:27:03,278 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin AutoSave
2017-01-11 14:27:03,281 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ImageReader
2017-01-11 14:27:03,283 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CuraProfileReader
2017-01-11 14:27:03,288 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin 3MFReader
2017-01-11 14:27:03,288 - WARNING - UM.PluginRegistry.loadPlugin [72]: Plugin CuraEngineBackend was already loaded
2017-01-11 14:27:03,291 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin 3MFWriter
2017-01-11 14:27:03,294 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin MachineSettingsAction
2017-01-11 14:27:03,299 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LayerView
2017-01-11 14:27:03,303 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin RemovableDriveOutputDevice
2017-01-11 14:27:03,350 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin XmlMaterialProfile
2017-01-11 14:27:03,351 - INFO - UM.Qt.QtApplication.__init__ [90]: Command line arguments: {'disable-textures': False, 'external-backend': False, 'file': []}
2017-01-11 14:27:03,357 - INFO - UM.VersionUpgradeManager.upgrade [100]: Looking for old configuration files to upgrade.
2017-01-11 14:27:03,359 - DEBUG - UM.Backend.Backend._logSocketState [167]: Socket state changed to Listening
2017-01-11 14:27:03,368 - INFO - UM.Backend.Backend.startEngine [71]: Started engine process: /usr/bin/CuraEngine
2017-01-11 14:27:03,369 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Calling engine with: ['/usr/bin/CuraEngine', 'connect', '127.0.0.1:49674', '-j', '/usr/share/cura/resources/definitions/fdmprinter.def.json', '']
2017-01-11 14:27:03,389 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-01-11 14:27:03,390 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Cura_SteamEngine version DEV
2017-01-11 14:27:03,390 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Copyright (C) 2014 David Braam
2017-01-11 14:27:03,391 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-01-11 14:27:03,392 - DEBUG - UM.Backend.Backend._logSocketState [171]: Socket state changed to Connected
2017-01-11 14:27:03,393 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] This program is free software: you can redistribute it and/or modify
2017-01-11 14:27:03,393 - DEBUG - UM.Backend.Backend._onSocketStateChanged [161]: Backend connected on port 49674
2017-01-11 14:27:03,393 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] it under the terms of the GNU Affero General Public License as published by
2017-01-11 14:27:03,393 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] the Free Software Foundation, either version 3 of the License, or
2017-01-11 14:27:03,399 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] (at your option) any later version.
2017-01-11 14:27:03,399 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-01-11 14:27:03,399 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] This program is distributed in the hope that it will be useful,
2017-01-11 14:27:03,400 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2017-01-11 14:27:03,400 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2017-01-11 14:27:03,400 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] GNU Affero General Public License for more details.
2017-01-11 14:27:03,401 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-01-11 14:27:03,401 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] You should have received a copy of the GNU Affero General Public License
2017-01-11 14:27:03,401 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2017-01-11 14:27:03,903 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [69]: Not attaching MachineSettingsAction to punchtec_connect_xl; Multi-extrusion printers are not supported
2017-01-11 14:27:05,922 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [69]: Not attaching MachineSettingsAction to ultimaker3; Multi-extrusion printers are not supported
2017-01-11 14:27:06,268 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [69]: Not attaching MachineSettingsAction to ultimaker3_extended; Multi-extrusion printers are not supported
2017-01-11 14:27:06,427 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [69]: Not attaching MachineSettingsAction to ultimaker_original_dual; Multi-extrusion printers are not supported
[... cut ...]
2017-01-11 14:27:20,811 - DEBUG - UM.Qt.Bindings.Theme.load [126]: Loading theme file: /usr/share/cura/resources/themes/cura/theme.json
2017-01-11 14:27:20,921 - DEBUG - UM.Qt.Bindings.Theme.load [186]: Loaded theme /usr/share/cura/resources/themes/cura
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
2017-01-11 14:27:21,023 - WARNING - cura.Settings.ContainerManager.getContainerMetaDataEntry [175]: Could not get metadata of container  because it was not found.
Segmentation fault (core dumped)
```

awhiemstra, 2017-01-11:
This is due to a bug in the Ubuntu Qt 5 packages. Please see https://github.com/Ultimaker/Cura/issues/1094 for more details.

closed 2017-01-11 by awhiemstra

grg2rsr, 2017-01-12:
thank you very much. The AppImage is functional workaround.