Ultimaker/Cura #1338: "machine_extruder_start_code" position in gcode

maukcc, 2017-01-12:
At the moment the sequence for starting a new extruder in gcode is:

1 switch extruder
2 set E to 0
3 "machine_extruder_start_code"
4 Heat up new extruder to "initial temp" and wait for it
5 prime
6 cool down old extruder
7 go to prime position
8 heat up new extruder to "print temp"
9 go to object or prime position
10 start printing
and other strange codes that are put in somewhere between1-10

The best sequence would be:

1 cool down old extruder
2 switch extruder
3 "machine_extruder_start_code"
4 go to object or prime position
5 start printing

This will require to set variables in "machine_extruder_start_code" as described in #1225
This way all machine vendors can put in their specific extruder change behavior.

discussion wanted ! on how to make it in a way no discussion is needed :)

maukcc, 2017-01-12:
Even swapping  position 3 and 4 in the current setup would be adequate for now.

awhiemstra, 2017-01-12:
There are a bunch of steps in the extruder switch behaviour that are quite deliberate. @BagelOrb can tell you more about it.

BagelOrb, 2017-01-13:
I don't really understand what the underlying issue is here.

Why do you want to heat up the extruder before the extruder start gcode?

You are condensing 10 steps into 5 steps, so there must be other changes which you want for some reason.

What else do you want and why?

maukcc, 2017-01-13:
The things that happen now (and I have no idea what some of them are meant to do) when starting a new extruder are probably very deliberately done for a specific machine. Unfortunately that specific machine is not our machine :)
Therefore I would like the program to do as less as possible, and let it be up to the machine maker to make the "start extruder code" into what they want.

The reason that we want the start code after the heatup code, is that we want to do some things with the extruder while it is hot, like prime and go to a specific position and wipe the nozzle in a specific way, and so on.

BagelOrb, 2017-01-13:
Ah that makes sense.

It's not really beneficial to our printers, though, but I will accept a pull request.

I'd imagine that `gcodeExport.cpp::switchExtruder(.)` will get an extra argument for the temperature to be met. In `gcodePlanner.cpp` the extra argument then has to be provided. Shouldn't be too difficult.

Ghostkeeper, 2017-01-22:
Note that cooling down the old extruder should still be done after heating up the new extruder, regardless of the printer model, since the prediction of the nozzle temperature for pre-heating depends on it.

maukcc, 2017-04-05:
Could someone point me in the direction where I would be able to change the sequence from

> <extruder_start_gcode>
> M109 Sxxxx

to
> M109 Sxxxx
> <extruder_start_gcode>

because that simple switch would solve all problems.
Unless someone can give a valid reason why they would want to do something (extruder_start_gcode) with the extruder while it is cold.

Ghostkeeper, 2017-04-05:
I'd expect the temperature commands to show up before the start g-code, unless there are temperature commands in the start g-code itself.

This is a sample of the beginning of g-code I just generated for a Custom FDM Printer:

    ;FLAVOR:RepRap
    ;TIME:617
    ;Filament used: 0.109933m
    ;Layer height: 0.1
    ;Generated with Cura_SteamEngine master
    M104 S200
    M109 S200
    G28 ;Home
    G1 Z15.0 F6000 ;Move the platform down 15mm
    ;Prime the extruder
    G92 E0
    G1 F200 E3
    G92 E0
    ;LAYER_COUNT:98
    ;LAYER:0
    M107
    G0 F3600 X39.623 Y39.349 Z0.3
    ;TYPE:SKIRT
    G1 F1800 X40.222 Y38.834 E0.01486
    G1 X40.87 Y38.383 E0.02971
    G1 X41.561 Y37.999 E0.04458

The start g-code starts with `G28 ;Home`, so as you can see the temperature commands are right before it.

maukcc, 2017-04-05:
This topic is about the extruder_start_gcode, not the machine_start_gcode :)

BagelOrb, 2017-04-05:
The start is written in `GCodeExport::startExtruder(int new_extruder)`
while the temperature command is written in
`LayerPlan::writeGCode(GCodeExport& gcode)` line 763.

It's not that simple to solve properly, so good luck!

maukcc, 2017-04-05:
If you say it is not that simple, then I believe you, and  am not even going to try :)

closed 2018-02-09 by maukcc