Ultimaker/Cura #1341: Cura 2.3.1 unable to import .ini from 15.04.6

juggleberry, 2017-01-12:
Unable to import .ini config files exported from 15.04.6 there just greyed out on macOS Sierra 10.12.2 and cant be selected
[image]

fieldOfView, 2017-01-13:
This is fixed kn 2.4. 

In 2.3, make sure to select "All supported filetypes (...)" in the filedialog.

juggleberry, 2017-01-13:
Thanks it was hidden on osx you have to select option in the dialog to see the options.
[image]

closed 2017-01-13 by juggleberry