Ultimaker/Cura #1342: High CPU usage when connected to um3 printer

peteruithoven, 2017-01-12:
I noticed quite a spike in cpu usage when the (connected) Ultimaker 3 is selected compared to a Ultimaker 2 for example. It sort of disturbs Cura's own UI but also other running apps. 
I'm running Cura 2.4 from the PPA on Elementary OS (which is build on Ubuntu 16.04) on a Dell XPS 15 (6th Gen Intel Core i7-6700HQ Quad Core (6M Cache, up to 3.5 GHz))
Have others noticed this?

nallath, 2017-01-13:
This could be that it's the camera stream. Is there are difference when viewing buildplate vs monitoring?

peteruithoven, 2017-01-13:
I'm also afraid it's the camera stream, my best guess is even that this started happening when the stream became live (instead of the images on a slower interval). I can't remember having this issue while it showed the images, even though I was on a way slower computer then. 

I'm not around the printer currently, so I can only test what happens when I open Cura without having the um3 connected. But then there is no issue at all, 0%cpu, 311MB RAM.

If there is any way for me to test this more thoroughly / measurable please let me know. 

(I love the live camera stream btw, so I hope we can figure out a solution)

peteruithoven, 2017-01-15:
So I tried this with my printer again, in combination with htop filtered to cura. Here's a comparison
Buildplate:
[image]
Monitoring:
[image]

fieldOfView, 2017-01-15:
Well, at least we know the mjpg stream is not being processed when the print-monitor is not open...

The "video" stream is handled in python. For all intends and purposes the stream is just a frame after frame of jpg files, one after the other. The network printing plugin receives the stream in a buffer and checks if it can find a start and end of a jpg file. If so it hands its that part of the buffer off to Qt to parse a single jpg. And so on. Obviously this is a bit of a hack and not a proper video player. Then again, one could argue that mjpg is not a proper video format ;-).

peteruithoven, 2017-01-15:
I think I get that, but running the same stream (http://192.168.1.114:8080/?action=stream) in Firefox or Chromium costs hardly any cpu at all.

fieldOfView, 2017-01-15:
I did not mean to imply that it is impossible to display an mjpg stream without high cpu load. 

Firefox and chromium contain and use  sophisticated media playback engines. Cura does not. It uses python to pick apart the video frame-by-frame.

Having said that, I don't see anywhere near your cpu load with the same(ish) code displaying a 640x480 mjpeg stream from an octoprint instance. I'll have a look-see if the UM3 code does something differently

awhiemstra, 2017-01-16:
Is this with master? Because there was a recent bug in master that caused an infinite amount of events to be sent on to Qt's event loop, eating up all of one core. That has been fixed at the end of last week, see https://github.com/Ultimaker/Uranium/commit/485c3033e68423031cdd07c9f3319250fc4bd7ac

peteruithoven, 2017-01-16:
Interesting! I'm using the [2.4 stable ppa](https://launchpad.net/~thopiekar/+archive/ubuntu/cura), which uses the last tagged version, which doesn't seem to include that commit yet. 
https://github.com/Ultimaker/Uranium/commits/2.4

awhiemstra, 2017-01-16:
It also does not include the commit that caused the issue in the first place, so I wonder what's going on there.

peteruithoven, 2017-03-10:
I'm still experiencing this issue using the [2.4 stable ppa](https://launchpad.net/~thopiekar/+archive/ubuntu/cura) and the [master ppa](https://launchpad.net/~thopiekar/+archive/ubuntu/cura-master). Is there progress on this? Have others been experiencing this issue?

Ghostkeeper, 2017-03-13:
Actually yes. I just tried this again and when no connection is present, Cura stays at 0.7% or so, very low. When a connection is present and I'm in the print monitor, it jumps to 330%.

Ghostkeeper, 2017-06-05:
Our product manager just removed this from our planning because the issue has been inactive for 3 months.

ChrisTerBeke, 2018-01-22:
The networking plugin has been completely rewritten, I think that'll fix this issue.