Ultimaker/Cura #1344: Tweak at Z should print a warning or error if it never activates (cura 2.4.0)

michaelni, 2017-01-13:
It seems tweak at Z behaves a bit unintuitive, for example setting the height to 0 causes it to never get enabled, i would have expected it to enable at the first layer (my intend was a simple temprature gradient for testing which covers the whole printed object)  setting the height to 0.5 works though. While this is IMHO a bug. Even if that is fixed one can mistakely enter silly things. The filter should warn the user that it never activated so the user knows of the issue and can correct it without inspecting gcode or waiting for the print to not go as planed

Ghostkeeper, 2017-01-22:
I'd also expect Tweak at Z to work at height 0. It probably has to do with how Tweak at Z looks at the Z-coordinate of a G0 or G1 move, tracking the Z until the requested Z is exceeded. And it starts at 0, so 0 never gets exceeded.

Ghostkeeper, 2018-03-16:
This is still a bug in the current master. Setting it to 0 doesn't work in ChangeAtZ. It causes that script to be ignored.

Ghostkeeper, 2018-05-18:
This was declined by our product owner. We don't think the feature is used often enough in combination with raft layers to fix this.

closed 2018-05-18 by Ghostkeeper