Ultimaker/Cura #1345: Tweak at Zs "number layers" should turn orange or red if its outside the object not at an arbitraray 50 layers (cura 2.4.0)

michaelni, 2017-01-13:
When i first used tweak at z i was surprised by the number layers option turning orange at an arbitrary layer count. its not clear why it does at that point, theres also no hint displayed why the value would be warned about

Ghostkeeper, 2017-01-22:
It should turn orange at a Z coordinate of 230: https://github.com/nallath/PostProcessingPlugin/blob/master/scripts/TweakAtZ.py#L78
This is because that would exceed the height of most printers. This treatment is a bit naive (it could use the actual printer height). But to get the actual printer height in there with inheritance would be quite difficult.

jackha, 2018-07-26:
deferred because of inactivity

closed 2018-11-26 by Appesteijn