Ultimaker/Cura #1346: Option to Hide Default Materials and Profiles

juggleberry, 2017-01-13:
Would it be possible to have the option hide default materials/ profiles that can't be edited so they don't display in the GUI?

Ghostkeeper, 2017-01-22:
This is the right place for feature requests. Thank you.

Hiding the default materials would also prevent users from creating their first non-default material, since the only way to create one is to clone an existing one. This is no problem for quality profiles though.

juggleberry, 2017-01-22:
My request was a bit ambiguous, i was thinking more along the lines of still having them in the settings menu but being able to hide them from the selection when choosing print settings
[image]

awhiemstra, 2017-01-23:
Ah, so more like being able to configure the displayed materials. This was one of the original ideas when we started doing materials stuff. It did not make it due to several constraints, but it is a good suggestion.

MiguelAngelLV, 2017-02-26:
+1 to idea.

I have many materials and the defaults profile are not useful for me

Ghostkeeper, 2017-03-01:
Status update: The feature is still in our backlog in our internal tracker. The product owner and change control board haven't come around to deciding on this issue yet.

Ghostkeeper, 2017-06-05:
Well now they have (at least, the product owner has; he's deciding alone nowadays). And he's decided to close this issue because it's been inactive for 3 months.

fieldOfView, 2017-06-10:
Having said that, Cura 2.6 will no longer show the default 2.85 mm materials when you have a 1.75 mm printer and vice-versa.

Ghostkeeper, 2018-09-17:
The original issue is implemented now in Cura 3.5 through the feature of making some materials favourite. The rest of the materials are then hidden under sub-menus.

LazaroFilm, 2019-02-09:
Hi, I might be really dumb, but how do you select your custom material profile to ad it to the favorite section? I see the favorite empty in the material manager but can't put anything in it.

Ghostkeeper, 2019-02-12:
In the material manager if you hover a material with your mouse, it'll show a star on the right hand side. If you click that star the material will be put in your favourites list.

Do you think it's too hidden? Maybe @Yian-Lai can think of something if they consider it a problem.