Ultimaker/Cura #1347: Custom profile ignores material settings

juggleberry, 2017-01-13:
When using a custom profile in cura 2.3.1 the settings from the material such as temp, bed temp and filament diamater are ignored by the profile.
[image]

Ghostkeeper, 2017-01-22:
Are you sure they aren't simply overwritten by that `.05 High Quality` profile you have there?

juggleberry, 2017-01-22:
Im not sure. How would i find out and if it is the case how would i go about resetting it so it doesn't override the the material settings?

Ghostkeeper, 2017-01-23:
Go to the profile manager (under the drop-down for quality profiles and then "Manage Profiles..."). When you select the profile, you should see on the right all the settings that it modifies.

nallath, 2017-07-27:
Closing this due to inactivity (I'm assuming the last response from ghostkeeper resolved the issue)

closed 2017-07-27 by nallath