Ultimaker/Cura #1349: Mouse Ears feature

probonopd, 2017-01-14:
It would be a welcome addition if there was an option in Cura to automatically switch on [Mouse Ears](https://ultimaker.com/en/community/17441-issues-with-detaching-brim-general-print-suggestions) for corners.

[image]

smartavionics, 2017-01-15:
One problem with DIY mouse ears is that the outer wall of the model goes around the mouse ear and so when the ear is removed the edge of the model where the ear joined is poor quality. What needs to happen is that it gets implemented the same as brim, i.e. the mouse ears butt against the perimeters which are not modified in any way and the ears can be removed with less damage to the printed part. So another way of thinking of mouse ears is "wide brim in selected locations".

Mearman, 2020-03-29:
Was this ever implemented?

fieldOfView, 2020-03-29:
Not as far as I know.

smartavionics, 2020-03-29:
But now we have the brim gap setting which can help make removing brim easier. Try a small gap, e.g. 0.1 or 0.2, works well for me.

nallath, 2023-05-10:
There are two plugins that can do this available on the marketplace:
https://marketplace.ultimaker.com/app/cura/plugins/5axes/TabPlus
https://marketplace.ultimaker.com/app/cura/plugins/5axes/TabAntiWarping

closed 2023-05-10 by nallath

MKlimuszka, 2024-05-08:
@nallath , neither of those plugins exist anymore. Can this be reopened?

gh13, 2024-05-08:
> @nallath , neither of those plugins exist anymore. Can this be reopened?

You could try downloading the curapackage directly from the dev's github [https://github.com/5axes?language=&page=1&q=&sort=name&tab=repositories](url) & install them manually

nicolasbuch, 2024-06-13:
This would be quite amazing to have as an alternative to the brim that goes all the way around. 

+1