Ultimaker/Cura #1351: Reset to defaults?

peteruithoven, 2017-01-15:
I'm trying to figure out the materials and profiles ui and wanted to reset everything to defaults, but I can't figure out how. There is a `Defaults` button in the preferences window, but in all pages except the General page it's unclickable. When I click it on the General page it resets the general settings, but it doesn't reset my materials and profiles. 
Is that feature still in development? If that's the case, why not hide the button?
Is there another way to reset my materials and profiles?

awhiemstra, 2017-01-16:
The quickest way is to remove everything in your local configuration directory. The (slightly more tedious) other option is to manually delete all custom materials, profiles and machines. It should not allow you to delete the default materials and profiles.

awhiemstra, 2017-01-16:
Also, the "Defaults" button in the preference window does nothing there because we never came up with what that should do. Having it restore materials or profiles to the default state is a rather destructive action that I am not sure we would want there.

peteruithoven, 2017-01-16:
Is there an easy way to discover this directory? (since it probably differs per os)
Maybe only show that button where appropriate then? It being disabled sort of gives the feedback that you're settings are the defaults.

awhiemstra, 2017-01-16:
Help -> Show Configuration Directory (yes, may not be the most logical place).

And I am not necessarily against the idea of using the defaults button to reset the profiles, but it would need a bit more thinking.

peteruithoven, 2017-01-16:
Thanks for all the quick answers @awhiemstra, I'll close this issue.

closed 2017-01-16 by peteruithoven

peteruithoven, 2017-01-25:
That folder (For me, on Linux `~/.config/cura`) doesn't seem to contain configuration of profiles and materials. It only contains .cura.cfg and this doesn't mention the custom materials and profiles I created. Are those stored somewhere else?

awhiemstra, 2017-01-25:
On Linux, we try to follow the XDG specification. This means preferences are stored in ~/.config/cura and local data is stored in ~/.local/share/cura . Machines etc. are considered local data and thus stored in ~/.local.