Ultimaker/Cura #1352: Duplicated material's profiles list is empty

peteruithoven, 2017-01-15:
To test this properly I would have liked to reset Cura to it's defaults but [I haven't figured out how or it's not possible yet](https://github.com/Ultimaker/Cura/issues/1084). 

I have selected the Ultimaker 3 as printer. I'm using 2.4.0 (from thopiekar's ppa).

I go to Manage materials I duplicate `PLA`. At this point I can activate the duplicate, view it's profiles in the preferences window and in the main window. But when I then change the `Display Name` and `Material Type` and I activate it, the profiles screen remains empty and the main window's Profile dropdown shows one `Not supported` option (besides the create, update, discard profile and manage profiles options). 
Basically I can't use my new Ultimaker to print with custom materials.

Probably related to: https://github.com/Ultimaker/Cura/issues/1238
My materials and profiles are saved, so it doesn't seem related to: https://github.com/Ultimaker/Cura/issues/1104

Below the logs of when I went through basically the following actions
1. Opened Cura
2. Went to Manage materials
3. Duplicated PLA material
4. Changed `Display Name` and `Material Type` to PLA2
5. Looking at profiles preferences (empty)
6. Shutdown Cura

Notice the many times the following error was logged: 
```
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
```

```
$ cura
Gtk-Message: Failed to load module "pantheon-filechooser-module"

(python3:28066): Gtk-WARNING **: Unable to locate theme engine in module_path: "pixmap",

(python3:28066): Gtk-WARNING **: Unable to locate theme engine in module_path: "pixmap",

(python3:28066): Gtk-WARNING **: Unable to locate theme engine in module_path: "pixmap",

(python3:28066): Gtk-WARNING **: Unable to locate theme engine in module_path: "pixmap",
2017-01-15 12:49:52,227 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ConsoleLogger
2017-01-15 12:49:52,235 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /usr/bin/CuraEngine
2017-01-15 12:49:52,236 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CuraEngineBackend
2017-01-15 12:49:52,238 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SimpleView
2017-01-15 12:49:52,239 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin WireframeView
2017-01-15 12:49:52,246 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin RotateTool
2017-01-15 12:49:52,249 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ScaleTool
2017-01-15 12:49:52,251 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin TranslateTool
2017-01-15 12:49:52,253 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CameraTool
2017-01-15 12:49:52,256 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin MirrorTool
2017-01-15 12:49:52,258 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SelectionTool
2017-01-15 12:49:52,259 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin OBJWriter
2017-01-15 12:49:52,260 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin OBJReader
2017-01-15 12:49:52,261 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin STLWriter
2017-01-15 12:49:52,265 - WARNING - STLReader.STLReader.<module> [26]: Could not find numpy-stl, falling back to slower code.
2017-01-15 12:49:52,265 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin STLReader
2017-01-15 12:49:52,267 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin FileLogger
2017-01-15 12:49:52,267 - WARNING - UM.PluginRegistry.loadPlugin [72]: Plugin ConsoleLogger was already loaded
2017-01-15 12:49:52,269 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LocalFileOutputDevice
2017-01-15 12:49:52,272 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CuraProfileWriter
2017-01-15 12:49:52,275 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin RemovableDriveOutputDevice
2017-01-15 12:49:52,277 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin GCodeProfileReader
2017-01-15 12:49:52,279 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SolidView
2017-01-15 12:49:52,281 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LegacyProfileReader
2017-01-15 12:49:52,286 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin 3MFReader
2017-01-15 12:49:52,332 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UM3NetworkPrinting
2017-01-15 12:49:52,332 - WARNING - UM.PluginRegistry.loadPlugin [72]: Plugin CuraEngineBackend was already loaded
2017-01-15 12:49:52,341 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin XmlMaterialProfile
2017-01-15 12:49:52,356 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin OctoPrintPlugin
2017-01-15 12:49:52,359 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin AutoSave
2017-01-15 12:49:52,361 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin PerObjectSettingsTool
2017-01-15 12:49:52,364 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ImageReader
2017-01-15 12:49:52,366 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin 3MFWriter
2017-01-15 12:49:52,370 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin XRayView
2017-01-15 12:49:52,395 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SliceInfoPlugin
2017-01-15 12:49:52,398 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LayerView
[... 739 lines cut ...]
2017-01-15 12:51:21,070 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Cura_SteamEngine version DEV
2017-01-15 12:51:21,071 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Copyright (C) 2014 David Braam
2017-01-15 12:51:21,071 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-01-15 12:51:21,072 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] This program is free software: you can redistribute it and/or modify
2017-01-15 12:51:21,072 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] it under the terms of the GNU Affero General Public License as published by
2017-01-15 12:51:21,073 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] the Free Software Foundation, either version 3 of the License, or
2017-01-15 12:51:21,073 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] (at your option) any later version.
2017-01-15 12:51:21,074 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-01-15 12:51:21,075 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] This program is distributed in the hope that it will be useful,
2017-01-15 12:51:21,075 - DEBUG - UM.Backend.Backend._logSocketState [171]: Socket state changed to Connected
2017-01-15 12:51:21,079 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2017-01-15 12:51:21,079 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2017-01-15 12:51:21,080 - DEBUG - UM.Backend.Backend._onSocketStateChanged [161]: Backend connected on port 49674
2017-01-15 12:51:21,080 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] GNU Affero General Public License for more details.
2017-01-15 12:51:21,081 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-01-15 12:51:21,081 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] You should have received a copy of the GNU Affero General Public License
2017-01-15 12:51:21,082 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2017-01-15 12:51:21,606 - DEBUG - CuraEngineBackend.CuraEngineBackend.slice [152]: Starting slice job...
2017-01-15 12:51:26,177 - WARNING - UM.Preferences.getValue [81]: Tried to get the value of non-existing setting info/automatic_update_check.
2017-01-15 12:51:29,166 - DEBUG - UM.Qt.QtApplication.windowClosed [198]: Shutting down cura
2017-01-15 12:51:33,878 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [197]: Attempting to kill the engine process
2017-01-15 12:51:33,878 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [203]: Killing engine process
2017-01-15 12:51:33,879 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [206]: Engine process is killed. Received return code -15
Segmentation fault (core dumped)
```

Ghostkeeper, 2017-01-22:
The QML is probably not designed to handle an empty list there, which is causing all those errors.

Regardless, the behaviour you described is correct. That is because you modified the `Material Type`. The quality profiles filter on this property. If the `Material Type` is modified it is considered a different type of material, sufficiently different that the profiles don't apply any more. And since you have no profiles any more, it shows up as "Not Supported".

peteruithoven, 2017-01-25:
Sounds like a solid explanation. But, when a material is duplicated, shouldn't it's profiles also be duplicated? 
Also, when I start with an empty list, from that window there is no way to add a profile. I need to go to the main Cura window, change something, store that as new profile. 
After that btw (even after restart), it's shown in the dropdown, but the Preferences > Profiles list is still empty. Is that a separate bug?

Ghostkeeper, 2017-01-26:
We don't duplicate all profiles along with a material, right now. You think that would be better? What about when a user wants to share a material with a new material type with "Export", should it share all its profiles as well?

Yeah a duplicate button for profiles in the profile manager would be nice.

If the profile drop-down is not in sync with the preferences > profiles list, that is also a bug. But indeed a separate one.

peteruithoven, 2017-02-02:
I'm not sure, I don't understand the architecture. I would like at least the basic fast, normal, good quality profiles when I create / duplicate a material. Currently there is no way, as far I can tell, to add custom materials because the profiles page stays empty.

peteruithoven, 2017-03-10:
I've just tried the [master ppa](https://launchpad.net/~thopiekar/+archive/ubuntu/cura-master), this seems fixed. Thanks!

closed 2017-03-10 by peteruithoven