Ultimaker/Cura #1353: Not retracting when moving between part(s) and prime tower

pouncingiguana, 2017-01-16:
When using dual extrusion with cura 2.3.1 or 2.4 beta, it is not retracting before moving to or from the prime tower. Tested with both versions on Linux and Windows 10.

Ghostkeeper, 2017-01-22:
Did you happen to have retraction disabled?
We have an open bug in our tracker that says "Tool retraction being done after moving to the switch (When retraction is disabled)".

pouncingiguana, 2017-01-23:
I do not have retraction disabled. It's retracting properly at all other places in the print, but not when it does the initial move away from the part towards the priming tower, or moves away from the priming tower on it's way back.

Ghostkeeper, 2017-01-24:
I can't reproduce this. For me it produces this g-code with a dual extrusion print when prime tower is enabled (I added comments):

    G1 F1800 X121.773 Y97.814 E157.81042 ;Last line of skin.
    M204 S2812
    M205 X18
    G1 F1500 E151.31042 ;Retracts here!
    G0 F10312.5 X124.885 Y99.854 ;Moving to switching bay.
    G0 X127.285 Y102.481
    G0 X213 Y189
    G1 F1200 E141.81042
    G92 E0
    
    T0 ;Extruder switch.
    G92 E0
    
    M109 S190
    G1 F1200 E9.5 ;Prime to retracted position.
    M104 T1 S100
    M106 S85
    G1 Z2.421
    G0 F10312.5 X167.5 Y186.5 Z2.42 ;Moving to prime tower.
    G1 Z0.42
    G0 X173.226 Y186.246
    M104 S200
    M204 S2000
    M205 X15
    ;TYPE:SUPPORT
    G1 F1500 E16 ;Unretract.
    G1 F1800 X174.216 Y187.236 E16 ;First line of prime tower.

pouncingiguana, 2017-01-26:
Hmmmm. We've been able to reproduce this over Windows and Linux (2.4 beta2). Only possible difference I can think of is that our printers are circular bed with 0,0 at center.

Ghostkeeper, 2017-01-27:
Perhaps it's an idea to export your project and upload it here so that we can reproduce it. It's under File -> Save Project. It saves the model along with all profiles, printers and settings that you're using. New feature of 2.4.

BagelOrb, 2017-03-08:
Perhaps you set the switch_extruder_retraction_amount at zero?

ianpaschal, 2018-02-27:
Closing for inactivity, irreproducibility, and the fact that we won't be making any changes fo versions below 2.7.

closed 2018-02-27 by ianpaschal