Ultimaker/Cura #1357: Does Cura 2.4.0 support dual extruders for custom printers?

JBKingdon, 2017-01-19:
Hi,
I was going to try the 2.4.0-BETA2 release but I can't seem to find any way of configuring for dual extruders. Am I missing something or is this not supported yet? 

Regards,
James

markwal, 2017-01-19:
2.3.1 has experimental dual support, but only for Ultimaker original dual.  It's actually possible to make it work with other printers, but not through the UI. You have to add some custom definition files in json format to the app files directly and since its a bit if a hack, guaranteed to break on upgrade.

Anyhow, you entered this issue on the OctoPrint upload plugin. Maybe you meant to enter it on the http://github.com/Ultimaker/Cura issue tracker?

JBKingdon, 2017-01-19:
That sounds interesting. The 2.4.0-Beta comes in some sort of launchable wrapper and I can't see where it's put the json files. Is there some way of unpacking it or should I just grab the current master from github (and go down the build dependencies rabbit hole...)

markwal, 2017-01-19:
What platform?

JBKingdon, 2017-01-19:
This is on Linux.

pouncingiguana, 2017-01-19:
If you're running ubuntu you can add the PPA to install a pretty current build of the beta (instructions below), then add you custom json files to /usr/share/cura/resources

sudo add-apt-repository ppa:thopiekar/cura
sudo apt update
sudo apt install cura

Ghostkeeper, 2017-01-22:
Also, the AppImage wrapper will unpack to a mounted image upon launching Cura. You can inspect the contents of the image while Cura is open if you can find where it lists the mounted images on your Linux distribution.

BagelOrb, 2017-03-08:
I think this question is answered.

closed 2017-03-08 by BagelOrb

probonopd, 2017-03-09:
> Also, the AppImage wrapper will unpack to a mounted image upon launching Cura. You can inspect the contents of the image while Cura is open if you can find where it lists the mounted images on your Linux distribution.

`/tmp/.mount_*` -- or you can just loop-mount it, since it is (also) an ISO file.

Patola, 2017-03-10:
I have been successfully using Cura with dual extrusion since 2.3, I even wrote a chapter in a book in my national language teaching how to do it. But you really have to edit the .def.json files... I used the ultimaker dual as a template