Ultimaker/Cura #1358: is "infill wipe" correct in what it does ?

maukcc, 2017-01-19:
[image]

As seen in the picture the infill goes up to the inner wall. The "infill wipe" goes the distance inside of the wall  (set rather large in the example).

I think the use of this infill wipe feature should be:
let the infill go towards the wall, minus the infill wipe distance. 
This way the end of the infill wipe distance ends up at the spot where the infill ends up in the current setup.
Only then will this feature have a function :)

Ghostkeeper, 2017-01-22:
Ah, it was designed to serve a different function.

It is meant to enhance adhesion between infill and wall. If you continue the movement of the infill lines on top of the wall, the infill line gets dragged/stretched a bit on top of the wall and solidifies there. This way they overlap a bit and therefore adhere better to each other, making the object stronger.

If we'd implement infill wipe as you describe it, you'd get neater retractions (not an issue in infill) and less material usage (a valid concern) but at the cost of lesser adhesion between wall and infill and therefore lesser strength.

maukcc, 2017-01-23:
I would think that "infill overlap" does the adhering to walls. And that this feature was to get rid of blobs forming where infill decelerates.

Ghostkeeper, 2017-01-23:
We sort of have two features with the same goal there: One with extrusion (infill overlap) and one without (infill wipe). Perhaps one of them should be re-purposed to fill the need of what you describe. I can't imagine it does a lot of good though, but perhaps it could improve the surface quality in case the wall is very thin.

BagelOrb, 2017-03-08:
The Infill Wipe setting is correct in what it does.
If you are sure you would like your functionality as a new feature, please make a new issue at github.com/Ultimaker/CuraEngine/issues

closed 2017-03-08 by BagelOrb