Ultimaker/Cura #1360: Skirt/Raft/Brim are affected by support structures

anton-veretenenko, 2017-01-19:
Application Version: 2.4.0-BETA2
Platform: Ubuntu 16.04 64bit

Steps to Reproduce:
Enable Support and Brim, Skirt or Raft.

Actual Results:
Brim, Skirt or Raft appears only where support structures are projected to the build plate plane.

Expected results:
Brim, Skirt or Raft should appear under actual object.

Additional Information:
Screen 1: Expected results, but support is disabled.
[image]

Screen 2: Support enabled, Brim is only under support structures and not actual object.
[image]

Screen 3: Support enabled, Skirt enabled, you can see that skirt is touching object now, but it should not.
[image]

Screen 4: Support enabled, Raft enabled, the same as with Brim.
[image]

Object used for screenshots: https://www.amazon.com/clouddrive/share/6W4ESxmwCGnI78zdxfbI6Rg4F0nLgLFiTEkptXzXcd0
It is rotated, but it does not matter.

anton-veretenenko, 2017-01-19:
Noticed that it bugs out only if initial layer height set to <=0.1mm and layer height is set to >=0.2mm.

Ghostkeeper, 2017-01-22:
CuraEngine removes any empty layers on the bottom of the print, and moves the object down so that it lays on the bed. If support is enabled, it will provide support beneath anything that floats, causing the layers to be no longer empty. It therefore doesn't move the mesh down any more and only draws support on the first layer where it needs support. The first layer gets encircled by the brim/skirt/raft. Since the first layer consists only of support, it only encircles that bit of support.

This is ultimately the result of your mesh not being entirely flat on the bottom. This causes Cura to think support is necessary in a few places but not all. Cura won't move the mesh down automatically because there are corners protruding on the bottom, so it thinks the object is resting on those corners. Only the corners are too small to print, so CuraEngine won't draw any walls there.

You can work around the problem by moving the mesh down slightly, like 0.15mm seems to work. Or you have to modify the mesh so that it has a smooth underside.

ianpaschal, 2018-02-27:
This is no longer an issue in 3.2.1.

closed 2018-02-27 by ianpaschal