Ultimaker/Cura #1361: While trying to add custom printer it silently fails when trying to find the stl

arthurlutz, 2017-01-19:
While trying to use https://github.com/DelphinPETER/Dagoma-Settings-For-Cura with Cura, I found the following problem : 

The documentation is not yet complete as to where resources can be customized, but I ended up loading them in ~/.local/share/cura/ and some of them appeared in cura (quality, materials). The printer definition itself was not showing up in the list of resources, so I copied that to /usr/share/cura/resources/definitions/ and so it showed up. But when adding it cura silently fails. 

In the log I get the following 

```
2017-01-19 22:19:37,388 - CRITICAL - cura.CrashHandler.show [32]: An uncaught exception has occurred!
2017-01-19 22:19:37,390 - CRITICAL - cura.CrashHandler.show [35]: Traceback (most recent call last):
2017-01-19 22:19:37,391 - CRITICAL - cura.CrashHandler.show [35]:   File "/usr/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 364, in addMachine
2017-01-19 22:19:37,391 - CRITICAL - cura.CrashHandler.show [35]:     Application.getInstance().setGlobalContainerStack(new_global_stack)
2017-01-19 22:19:37,392 - CRITICAL - cura.CrashHandler.show [35]:   File "/usr/lib/python3/dist-packages/UM/Application.py", line 136, in setGlobalContainerStack
2017-01-19 22:19:37,393 - CRITICAL - cura.CrashHandler.show [35]:     self.globalContainerStackChanged.emit()
2017-01-19 22:19:37,393 - CRITICAL - cura.CrashHandler.show [35]:   File "/usr/lib/python3/dist-packages/UM/Signal.py", line 150, in emit
2017-01-19 22:19:37,394 - CRITICAL - cura.CrashHandler.show [35]:     func(dest, *args, **kwargs)
2017-01-19 22:19:37,394 - CRITICAL - cura.CrashHandler.show [35]:   File "/usr/lib/python3/dist-packages/UM/Scene/Platform.py", line 52, in _onGlobalContainerStackChanged
2017-01-19 22:19:37,395 - CRITICAL - cura.CrashHandler.show [35]:     path = Resources.getPath(Resources.Meshes, mesh_file)
2017-01-19 22:19:37,395 - CRITICAL - cura.CrashHandler.show [35]:   File "/usr/lib/python3/dist-packages/UM/Resources.py", line 70, in getPath
2017-01-19 22:19:37,396 - CRITICAL - cura.CrashHandler.show [35]:     raise FileNotFoundError("Could not find resource {0} in {1}".format(args, resource_type))
2017-01-19 22:19:37,396 - CRITICAL - cura.CrashHandler.show [35]: FileNotFoundError: Could not find resource ('dagoma_discoeasy200_plateform.stl',) in 3
```
Indeed, I can fix that by copying the meshes/dagoma_discoeasy200_plateform.stl to /usr/share/cura/resources/meshes

This request would be to display the error in the user interface so one can find a fix. 

Other than that I would argue that not being able to load the stl file that is eye candy could be a warning but this make that custom printer usable...

Ghostkeeper, 2017-01-22:
Problem with showing that error in the GUI: The error occurs before the GUI is even started.

arthurlutz, 2017-01-23:
@Ghostkeeper when the error occurs the cura window is open. And even if it was, I'm sure there can be a mechanism for reading an error log and displaying the last errors to the user

Ghostkeeper, 2017-07-10:
Our product manager just closed this for being inactive for 3 months.

jackha, 2018-07-30:
Currently we have an error dialog that also catches problems during startup.

closed 2018-07-30 by jackha