Ultimaker/Cura #1363: 3MF v1.1 is here!

rogersachan, 2017-01-21:
See http://3mf.io/3mf-version-1-1-is-here/ for the details, but all in all nothing major has changed, so 1.0 and 1.1 can be used interchangably,

Changes are as follows:

* Page 13: Restriction to absolute paths within the package. This was already true in the XSD schema, but is now reminded in the text.
* Page 15: Restricted XML content to UTF-8 to simplify parsers. We added a reminder of how XML namespace inheritance works.
* Pages 25-28: Added object type “solidsupport,” which like a model, must be manifold (hence solid). Objects of type “support” are not intended as solid objects, but as thin walls (regardless of whether they are manifold or not) to be manufactured at the machine’s precision. This ensures no ambiguity in whether support structures should be filled or not.
* Page 31: Restriction that base materials cannot form gradients, in order to ensure that splitting edges in a core-spec 3MF still results in a core-spec 3MF, rather than requiring composite materials from the materials and properties extension.
* Page 46: The only change to the XSD schema is to add “solidsupport.”

Ghostkeeper, 2017-01-22:
The only change that Cura could implement is that new "solidsupport" feature. But since Cura currently doesn't support similar features either such as the gradient infill stuff, I don't think there is much haste to implement that.

rogersachan, 2017-01-22:
See http://3mf.io/wp-content/uploads/2016/03/3MFcoreSpec_1.1.pdf for the document, since the link in page is broken.

superusercode, 2017-07-27:
@nallath Was the feature added?

Ghostkeeper, 2017-08-10:
No, we don't have solidsupport.

rogersachan, 2018-05-09:
Did this ever land?

Ghostkeeper, 2018-06-08:
No, we still don't have solidsupport-support. We have the feature in Cura but it's not automatically enabled when you load a 3MF file in Cura.

The gradient infill is in a side track under development though (but that will take a while still).

rogersachan, 2019-04-24:
Is there a tracking issue for solidsupport-support?

Ghostkeeper, 2019-04-25:
Yes, this one.

GregValiant, 2025-07-03:
I'll go ahead and close this.  If you feel strongly about it then it can be reopened.

closed 2025-07-03 by GregValiant