Ultimaker/Cura #1364: (Feature) Add Renderer colors to theme.json

rudowinger, 2017-01-21:
As a friend of "dark" 3D software, it would be great if the theme.json is extended by _renderer_ color options like background-color

[image]

fieldOfView, 2017-01-21:
I looked into this a little while ago. The tricky part is that the theme layer is used in the part of the UI that is specified in QML, but the viewport is handled by the underlying python layer. Python currently has no access to the theme singleton.

rudowinger, 2017-01-21:
Maybe a Python level "dark_mode" setting then, that inverts the colors?

fieldOfView, 2017-01-22:
I have found a way to access the theme from the Python side of things, and I am working on a set of Pull Requests to make the viewport colors themable.

fieldOfView, 2017-01-23:
See https://github.com/Ultimaker/Cura/pull/1370 and https://github.com/Ultimaker/Uranium/pull/215

closed 2017-01-25 by jackha