Ultimaker/Cura #1365: COmpilation error.

rewolff, 2017-01-22:
Hi, 

I'm on the latest stable Ubuntu: 16.04, and I tried compiling cura from github. 

Following the instructions on this repository's readme, I got this far (after upgrading "protobuf": 

/home/wolff/3d/cura/cura-git/cura-build/build/CuraEngine-prefix/src/CuraEngine-build/Cura.pb.h: In member function ‘const string& cura::proto::GCodePrefix::data() const’:
/home/wolff/3d/cura/cura-git/cura-build/build/CuraEngine-prefix/src/CuraEngine-build/Cura.pb.h:3270:27: error: no matching function for call to ‘google::protobuf::internal::ArenaStringPtr::GetNoArena() const’
   return data_.GetNoArena();
                           ^
In file included from /home/wolff/3d/cura/cura-git/cura-build/build/CuraEngine-prefix/src/CuraEngine-build/Cura.pb.h:23:0,
                 from /home/wolff/3d/cura/cura-git/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/commandSocket.h:13,
                 from /home/wolff/3d/cura/cura-git/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/gcodeExport.h:14,
                 from /home/wolff/3d/cura/cura-git/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/gcodePlanner.h:7,
                 from /home/wolff/3d/cura/cura-git/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/sliceDataStorage.h:11,
                 from /home/wolff/3d/cura/cura-git/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/bridge.cpp:4:
/home/wolff/3d/cura/cura-git/cura-build/build/inst/include/google/protobuf/arenastring.h:219:31: note: candidate: const string& google::protobuf::internal::ArenaStringPtr::GetNoArena(const string*) const
   inline const ::std::string& GetNoArena(const ::std::string* /* default_value */) const {
                               ^
/home/wolff/3d/cura/cura-git/cura-build/build/inst/include/google/protobuf/arenastring.h:219:31: note:   candidate expects 1 argument, 0 provided
CMakeFiles/_CuraEngine.dir/build.make:70: recipe for target 'CMakeFiles/_CuraEngine.dir/src/bridge.cpp.o' failed
make[5]: *** [CMakeFiles/_CuraEngine.dir/src/bridge.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/_CuraEngine.dir/all' failed
make[4]: *** [CMakeFiles/_CuraEngine.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/CuraEngine.dir/build.make:111: recipe for target 'CuraEngine-prefix/src/CuraEngine-stamp/CuraEngine-build' failed
make[2]: *** [CuraEngine-prefix/src/CuraEngine-stamp/CuraEngine-build] Error 2
CMakeFiles/Makefile2:549: recipe for target 'CMakeFiles/CuraEngine.dir/all' failed
make[1]: *** [CMakeFiles/CuraEngine.dir/all] Error 2
Makefile:105: recipe for target 'all' failed

This leaves me stumped: I know too little C++ to see what's wrong, or what might help to fix this. 
As far as I can see the code is simply wrong.

rewolff, 2017-01-22:
P.S. I the tried downloading the packaged .deb file which seems to have overwritten my existing older version of CURA that worked quite well for my own printer. Because I wanted to slice for a friends delta too I wanted to try upgrading. Anyway, the old version is now gone. I can no longer slice.

BerndJM, 2017-01-22:
If you are not in the need (for whatever reason) to compile it by yourself you could use thopiekar's PPA:
https://launchpad.net/~thopiekar/+archive/ubuntu/cura

rewolff, 2017-01-22:
Thanks! Tried that.... 
```

cura.CrashHandler.show [32]: An uncaught exception has occurred!
cura.CrashHandler.show [35]: Traceback (most recent call last):
cura.CrashHandler.show [35]:   File "/usr/bin/cura", line 47, in <module>
cura.CrashHandler.show [35]:     import Arcus #@UnusedImport
cura.CrashHandler.show [35]: RuntimeError: the sip module implements API v11.0 to v11.2 but the Arcus module requires API v11.3

```

Ghostkeeper, 2017-01-22:
Do you know what version of Protobuf you used? The bug seems to be in some of their files, rather than Cura's.

rewolff, 2017-01-23:
I cloned the GIT repository, compiled and installed it. I see releases in the 3.2.0rc2 range, but there are twenty different release files there and I haven't the foggiest idea which one I should download and install.

Ghostkeeper, 2017-01-23:
You probably used master then, which is currently the 3.2.0rc2 release because they use the convention that development doesn't happen on the master branch but gets merged in there after their testing phase.

I've been using version 3.0.0. Perhaps you can try that one: https://github.com/google/protobuf/releases/tag/v3.0.0

rewolff, 2017-01-23:
It stopped quickly when I retried. Then I did "make clean" and tried again. It's now running much longer.... Getting hopeful. 
The trick to going to 3.0.0 (I'm not good with git, but trying to learn) was: git checkout -b v300 v3.0.0

rewolff, 2017-01-23:
It now stops at:
```
[ 51%] Performing install step for 'Qt'
strip:/home/wolff/3d/cura/cura-git/cura-build/build/inst/bin/syncqt.pl: File format not recognized

```
where I have to agree with "strip" in that th eperl scirpt cannot be stripepd. 

(Excluse my typing: rdesktop ovre a slow link makes me type bad....)

Ghostkeeper, 2017-01-24:
> P.S. I consider this sloppy code: lots and lots of compiler warnings.

Yeah there's still a lot of unused parameters. I'm removing them when I touch the function but there is some code that we're just not touching for a very long time. Perhaps we should schedule a refactor to remove them.

Ghostkeeper, 2017-01-24:
I haven't seen that Perl script before and I wouldn't know why Qt even tries to compile that...

awhiemstra, 2017-01-24:
If you are on Ubuntu and using cura-build, you may want to try the environment-split branch. This does not try to build the entire environment, but just our main components. It assumes the environment is already set up, though it does perform checks to make sure all dependencies are available. This should give a much cleaner build process.

rewolff, 2017-01-28:
I switched to the PPA build of cura. It crashes with: 

```
2017-01-28 13:10:22,407 - DEBUG - CuraEngineBackend.CuraEngineBackend.slice [156]: Starting slice job...
Segmentation fault (core dumped)
```

Although I've tried Slic3r in the past, I liked cura better. But having been without a working slicer for a week, I decided to try Slic3r again. I like it much better already. For one, the distribution file simply works. 

Stuff like separate settings for first layer extruder and bed temperature are useful to create good prints. 
Goodbye and thanks for all the fish.

Ghostkeeper, 2017-01-30:
Good luck! Sorry Cura didn't work out for you here. That PPA is not officially supported.

In version 2.4, we switched to a different distribution method for Linux (AppImage rather than .deb), and added settings for first layer bed & extruder temperature. We hope it might lure you back next week.

closed 2017-07-27 by nallath