Ultimaker/Cura #1366: Failed to load STL file which loads in Cura 15.04.6

Jastonite, 2017-01-22:
The attached file failed to load in Cura 2.3.1, but loads and slices in Cura 15.04.6.  It is a file exported from Blender 2.78a

[rhino.stl.txt](https://github.com/Ultimaker/Cura/files/722147/rhino.stl.txt)

Ghostkeeper, 2017-01-22:
It seems to produce an error in QHull:
    2017-01-23 00:29:26,848 - DEBUG - UM.Mesh.MeshData.calculateNormalsFromVertices [379]: Calculating normals took 0.01911330223083496 seconds
    2017-01-23 00:29:26,868 - DEBUG - STLReader.STLReader.read [63]: Loaded a mesh with 257400 vertices
    QH6154 qhull precision error: initial facet 1 is coplanar with the interior point
    ERRONEOUS FACET:
    - f1
        - flags: bottom simplicial flipped
        - normal:         0       -1        0
        - offset:         -0
        - vertices: p1(v2) p5(v1) p0(v0)
        - neighboring facets: f2 f3 f4

    While executing:  | qhull i Qt
    Options selected for Qhull 2012.1 2012/02/18:
      run-id 339934082  incidence  Qtriangulate  _pre-merge  _zero-centrum
      _max-width 0,12  Error-roundoff 1,7e-16  _one-merge 1,2e-15
      _near-inside 6,1e-15  Visible-distance 3,5e-16  U-coplanar-distance 3,5e-16
      Width-outside 6,9e-16  _wide-facet 2,1e-15

    The input to qhull appears to be less than 3 dimensional, or a
    computation has overflowed.

    Qhull could not construct a clearly convex simplex from points:
    - p2(v3):     0     0    -0
    - p1(v2):     0     0  0,12
    - p5(v1):  0,12     0     0
    - p0(v0):     0     0     0

    The center point is coplanar with a facet, or a vertex is coplanar
    with a neighboring facet.  The maximum round off error for
    computing distances is 1,7e-16.  The center point, facets and distances
    to the center point are as follows:

    center point  0,03125        0  0,03125

    facet p1 p5 p0 distance=    0
    facet p2 p5 p0 distance=    0
    facet p2 p1 p0 distance= -0,031
    facet p2 p1 p5 distance=    0

    These points either have a maximum or minimum x-coordinate, or
    they maximize the determinant for k coordinates.  Trial points
    are first selected from points that maximize a coordinate.

    The min and max coordinates for each dimension are:
      0:         0     0,125  difference= 0,125
      1:         0         0  difference=    0
      2:         0     0,125  difference= 0,125

    If the input should be full dimensional, you have several options that
    may determine an initial simplex:
      - use 'QJ'  to joggle the input and make it full dimensional
      - use 'QbB' to scale the points to the unit cube
      - use 'QR0' to randomly rotate the input for different maximum points
      - use 'Qs'  to search all points for the initial simplex
      - use 'En'  to specify a maximum roundoff error less than 1,7e-16.
      - trace execution with 'T3' to see the determinant for each point.

    If the input is lower dimensional:
      - use 'QJ' to joggle the input and make it full dimensional
      - use 'Qbk:0Bk:0' to delete coordinate k from the input.  You should
        pick the coordinate with the least range.  The hull will have the
        correct topology.
      - determine the flat containing the points, rotate the points
        into a coordinate plane, and delete the other coordinates.
      - add one or more points to make the input full dimensional.
    2017-01-23 00:29:26,965 - CRITICAL - cura.CrashHandler.show [32]: An uncaught exception has occurred!
    2017-01-23 00:29:26,966 - CRITICAL - cura.CrashHandler.show [35]: Traceback (most recent call last):
    2017-01-23 00:29:26,966 - CRITICAL - cura.CrashHandler.show [35]:   File "/home/ruben/Projects/Cura/cura/CuraApplication.py", line 494, in event
    2017-01-23 00:29:26,966 - CRITICAL - cura.CrashHandler.show [35]:     return super().event(event)
[... 91 lines cut ...]
        correct topology.
      - determine the flat containing the points, rotate the points
        into a coordinate plane, and delete the other coordinates.
      - add one or more points to make the input full dimensional.
    2017-01-23 00:29:27,038 - ERROR - UM.Logger.logException [65]: Exception: Exception occurred while loading file /home/ruben/Downloads/rhino.stl
    2017-01-23 00:29:27,039 - ERROR - UM.Logger.logException [69]: Traceback (most recent call last):
    2017-01-23 00:29:27,040 - ERROR - UM.Logger.logException [69]:   File "/home/ruben/Projects/Uranium/UM/FileHandler/ReadFileJob.py", line 57, in run
    2017-01-23 00:29:27,040 - ERROR - UM.Logger.logException [69]:     self.setResult(self._handler.readerRead(reader, self._filename))
    2017-01-23 00:29:27,040 - ERROR - UM.Logger.logException [69]:   File "/home/ruben/Projects/Uranium/UM/Mesh/MeshFileHandler.py", line 37, in readerRead
    2017-01-23 00:29:27,040 - ERROR - UM.Logger.logException [69]:     extents = result.getMeshData().getExtents()
    2017-01-23 00:29:27,041 - ERROR - UM.Logger.logException [69]:   File "/home/ruben/Projects/Uranium/UM/Mesh/MeshData.py", line 162, in getExtents
    2017-01-23 00:29:27,041 - ERROR - UM.Logger.logException [69]:     data = numpy.pad(self.getConvexHullVertices(), ((0, 0), (0, 1)), "constant", constant_values=(0.0, 1.0))
    2017-01-23 00:29:27,041 - ERROR - UM.Logger.logException [69]:   File "/home/ruben/Projects/Uranium/UM/Mesh/MeshData.py", line 237, in getConvexHullVertices
    2017-01-23 00:29:27,041 - ERROR - UM.Logger.logException [69]:     convex_hull = self.getConvexHull()
    2017-01-23 00:29:27,042 - ERROR - UM.Logger.logException [69]:   File "/home/ruben/Projects/Uranium/UM/Mesh/MeshData.py", line 229, in getConvexHull
    2017-01-23 00:29:27,053 - ERROR - UM.Logger.logException [69]:     self._computeConvexHull()
[... cut ...]
    2017-01-23 00:29:27,055 - ERROR - UM.Logger.logException [69]:   File "scipy/spatial/qhull.pyx", line 2266, in scipy.spatial.qhull.ConvexHull.__init__ (scipy/spatial/qhull.c:21773)
    2017-01-23 00:29:27,055 - ERROR - UM.Logger.logException [69]:   File "scipy/spatial/qhull.pyx", line 338, in scipy.spatial.qhull._Qhull.__init__ (scipy/spatial/qhull.c:4039)
    2017-01-23 00:29:27,055 - ERROR - UM.Logger.logException [69]: scipy.spatial.qhull.QhullError: Qhull error

File loading should not break on this, in my opinion.

Ghostkeeper, 2017-06-05:
Our product manager just removed this issue from our tracker for being inactive for 3 months.

awhiemstra, 2017-06-06:
Note also that we did some fixes to this recently, so things should be a bit better with regards to this error.

Ghostkeeper, 2017-07-10:
Well no, it isn't. Now Cura actually crashes hard instead of just giving an error.

ianpaschal, 2018-02-21:
Closing due to age.

closed 2018-02-21 by ianpaschal

Ghostkeeper, 2018-03-13:
I can't figure out what fixed it, but the model no longer causes Cura to crash.