Ultimaker/Cura #1369: wrong %userprofile% folder with itinerant profile (W7)

ljoets, 2017-01-23:
Hello,
I have to deploy latest stable Cura ( version 2.3.1) on W7x64 computers in a school.
Users have itinerant windows profiles on a sambaedu3 server.
In theory, %userprofile% is on a network drive. And I expected to find .cura folder on it, but .cura folder stays on C:\Users
Is there a method to specify %userprofile% at launch ?
Thanks.
I had this profil problem with precedent version, and found a workaround #(505) but cfg files goes to C:\users and I can't find profile.py in C:\Programs\Cura 2.3

ljoets, 2017-01-24:
I found here where the windows userprofile is defined : 
https://github.com/Ultimaker/Cura/blob/d391639b48e3e3adc3af6c935e7595490ed4025a/cura_app.py
But I can't find this file int the path of an installed version of Cura.
How can I change that file after install ?
Thanks

fieldOfView, 2017-01-24:
Unfortunately this cannot be easily changed with the official releases, as the relevant code is distributed in a compiled form. 

This needs to be changed in the source, in two places actually. I'll prepare a pull request so it gets fixed in Cura 2.5. At this point it is unlikely the fix is going to make it in the 2.4 release.

Technical analysis: Cura does try to do the right thing by asking Python to expand the "~" directory to the user profile folder. This should work, but apparently there are situations where it doesn't. Instead of ```os.path.expanduser("~/AppData/Local/cura/")``` we should probably use ```os.path.expandvars("%LOCALAPPDATA%\cura")```. That way we ask what the OS says, instead of Python.

awhiemstra, 2017-01-24:
This is what the Python documentation says about expanduser:

> On Windows, HOME and USERPROFILE will be used if set, otherwise a combination of HOMEPATH and HOMEDRIVE will be used. An initial ~user is handled by stripping the last directory component from the created user path derived above.

So if that does not actually work, then there is a bug in Python. What does the %USERPROFILE% variable contain for you?

fieldOfView, 2017-01-24:
Regardless of what USERPROFILE returns, it is better to use the LOCALAPPDATA variable, because a user can change the path of only the local appdata folder.

ljoets, 2017-01-24:
Ok, thanks.
On computers I manage in our school (windows 7x64 in a domain),
echo %userprofile% -> C:\Users\<my_login>
echo %appadata% -> K:\profil (it is what I need)
echo %localappdata% -> C:\Users\<my_login>\AppData\Local
When will Cura 2.5 be ready ? ;-)

Ghostkeeper, 2017-01-24:
Cura 2.5 will take a few months. We have no deadlines yet but the original aim was to release a version every 2 months. In practice it's a bit slower than that.

fieldOfView, 2017-01-24:
Ah, so the real problem is you want Cura to store its data in the Roaming appdata folder instead of the Local appdata folder. How do you do this with other applications? Most applications on my computers store their data in the Local appdata.

ljoets, 2017-01-24:
I don't know...
It's just working ;-)

fieldOfView, 2017-01-24:
As a workaround, you could create a junction C:\Users<my_login>\AppData\Local\cura to K:\profil\cura after installing Cura.
https://en.wikipedia.org/wiki/NTFS_junction_point

Ghostkeeper, 2017-03-23:
Hopefully this was fixed by https://github.com/Ultimaker/Uranium/pull/216

The fix would be included in version 2.6 (not the 2.5 beta that's going out today). Unless you can create a Windows build of your own, you'll be able to try it out in June roundabout.

fieldOfView, 2017-03-23:
No, it is not (as is mentioned in the pr comments). Cura really had to use the AppData/Roaming instead of AppData/Local folder. 

I'll see if I can make a follow-up pr. That PR  would then have to include an update/migrate step to move the old folder from local too roaming.

closed 2018-02-27 by ianpaschal