Ultimaker/Cura #1371: Colors not defined + Colors with inconvenient connection in theme.js?

rudowinger, 2017-01-23:
Hey,

While working on a "dark mode" adobeésque theme for Cura, I stumbled on a few hickups that I want to report:

For a large part, the theming works very good and Cura can be enhanced very well with using the theme subsystem!!!

Here you can see a screenshot of my dark theme in progress:
[image]

... but as you can see, some colors have inconvenient connections or cannot be customized at all.

I made a complete-red theme to check for un-themeable parts:
[image]

The issues I found: (maybe some things come from a plugin and use system wide values such as primary color? Help in that case is very much appreciated!)

Unthemeable parts:
- Screen "Print Setup - Recommended": Background field on "infill", Background field on "Enable Support, Build Plate adhesion", Background of "Need help improving..."
- Background of Slicer Status Window
- Tooltip of "View - Layers" slider

Would be great to have independent theme options:
- "3D view - Layers" tooltip background and caption text
- "Slicer Status" text caption field, i.e. "Slicing...", Button "Save To file"
- "3D view" - Title, Metrics, Time & Filament Stats

Hope, my feedback helps, thank you for reading :)

fieldOfView, 2017-01-23:
This PR removes the white blocks:
https://github.com/Ultimaker/Cura/issues/1372

fieldOfView, 2017-01-23:
Forgot to say: that dark theme looks good!

rudowinger, 2017-01-24:
Thanks (also for your quick Pull requests!) You can have it, when it's done

closed 2017-01-26 by Ghostkeeper

fieldOfView, 2017-03-30:
@rudowinger have you been able to check the 2.5 beta? It contains the fixes to the theme system you need to fix your dark theme.

rudowinger, 2017-03-31:
@fieldOfView So it is in Beta? Was too dumb to compile - thanks for letting me know - will look into it!

fieldOfView, 2017-04-01:
Yes, the theme fixes are in the 2.5 beta.
https://ultimaker.com/en/blog/39116-introducing-cura-25-beta

Ofcourse it is entirely possible that some new issues have cropped up with the theme, so it would be great if you could give it a go before 2.5 goes out of beta.

fieldOfView, 2017-06-19:
Cura 2.7 will have a proper theme selection, and more importantly it has "inheritance" for themes. That means that you can make a theme that changes just the colors (or even just a hand-full of colors). Themes can also be loaded from the configuration folder, so you don't even need admin privileges to edit the theme.

A very minimal theme that only changes the background of the sidebar to read could be as simple as this theme.json file in a new folder inside the themes folder:
'''
{
    "metadata": {
        "name": "Red Sidebar",
        "inherits": "cura"
    },
    "colors": {
        "sidebar": [255, 0, 0, 255]
    }
}
'''

Would be nice to include that dark theme in Cura 2.7. Can I help you with that @rudowinger?

rudowinger, 2017-08-02:
Oh, just've seen it right now :/
As seen from the "Add a dark theme" pull request, you did the same as me, 
defining two base shades of gray and the Cura blue as highlighting colors, as well as white for most of the texts... only one thing is missing to complete the theme - 
the darkest shade of gray for floating info texts and tooltips: #282828
Thank you, @fieldOfView!