Ultimaker/Cura #1374: layers turned floating after rotated 90 degrees

kitopitodave, 2017-01-24:
I'm having an issue with parts of my model becoming empty layers when rotated 90 degrees to its side. The issue goes away if I removed the pole from the model or have the pole pointing up instead of on the side (but will need to print supporting structures, which adds another 2hrs). 

I've attached the stl files.  Is this a cura issue or sketchup issue? 

Cura 2.3.1 for Mac
The model is drawn using sketchup 2017 mac version.

[image]
Model sitting in the orientation as designed. No issue near pole area

[image]
model turned onto its side for faster printing without supporting structure. However, parts of the model turned empty near the pole area. 

[image]
No issue if the pole is removed. 

[Archive.zip](https://github.com/Ultimaker/Cura/files/726044/Archive.zip)

fieldOfView, 2017-01-24:
If you change the view mode to "XRay", you will see red areas in your model with poles. This means there is an error in your model, making the slicing result unpredictable. You *might* have luck turning on/off some of the "Mesh Fixes" in Cura.

Sketchup is not the best modeler to create models for 3d printing; it just was not designed with 3d printing in mind. I'm not saying it cannot be done, but you will run into issues like this.

kitopitodave, 2017-01-24:
Great, thanks for the reply Aldo. I'll check on that, and maybe consider
other design software.
best.

Dave

closed 2017-01-24 by Ghostkeeper