Ultimaker/Cura #1376: opening all previously opened files until restart

spotdogg, 2017-01-24:
open model 1....clear build plate or delete model. Open model2 and both the model1 and model2  open. Clear or delete again.  Then opening model3 results in model 1,2 and 3 opening. Clear again open model4 results in 1,2,3 and 4 opening.  2.4.0 from repo on 16.10. x920-8core-16Gb-7970GPU

awhiemstra, 2017-01-25:
Sounds like you are using KDE. There is a known bug in the KDE file dialog that causes this. See https://bugs.kde.org/show_bug.cgi?id=350408

Ghostkeeper, 2017-01-26:
Can't reproduce this using Unity on Ubuntu 16.10.

spotdogg, 2017-01-29:
Sorry Ghosty, Many Thanks awhiemstra. Exactly right.  A quick fling with other desktops proves you right. Its a KDE issue.  A bit sad to see it reported in 2015 and still not fixed. Bit of a strange one as now I know what it is I have tried hard to make other apps behave the same without success?

closed 2017-01-29 by spotdogg

Ghostkeeper, 2017-01-30:
There's many different types of file dialogue. Perhaps we are using the one type that happens to cause problems on KDE.

If we can detect KDE it should be possible to switch to a file dialogue that disallows multiple files just for KDE. Could be a temporary patch.