Ultimaker/Cura #1377: Cura 2.4.0-Beta 2 g-code sets hotend temp to 255C when the temp is set to 250C in the GUI

joochung, 2017-01-25:
So I have a i3 profiled printer in Cura 2.4.0-Beta2. It was inherited from Cura 2.3.1. I have the printing Temperature set to 250C but the g-code that gets created includes the following commands at the beginning:

```
    M104 S255
    M109 S255
```

It sets the bed temp properly but not the hotend temp.

Ghostkeeper, 2017-01-25:
In Cura 2.4 there is a new setting, Initial Layer Printing Temperature, controlling the temperature of the hot end for just the duration of the first layer. This setting defaults to 5 degrees hotter than the normal printing temperature. You can change it to 250 degrees if you want, but it helps a bit with bed adhesion to print slightly hotter in the first layer.

Ghostkeeper, 2017-01-25:
Duplicate of #1315.

closed 2017-01-25 by Ghostkeeper