Ultimaker/Cura #1380: Seperate walls printed at the same time

dk95, 2017-01-25:
If there are independant walls on one layer (e.g. a plate with holes), cura currently prints each wall line seperately. Starting with the innermost, ending at the outer wall line. This results in many retractions&travel moves as the printhead jumps from each independent wall part to the next one on each wall line. And especially on the first layer the chance increases that walls of a smaller/detailed part come loose due to the many travel moves.
Would it be possible to add a feature that prints independent wall parts (e.g. for a hole) at once?

Ghostkeeper, 2017-01-26:
We've heard this feature request before, but I can't find the duplicate of this at the moment so I'll leave it open. It's a more common request.

dk95, 2017-01-26:
Thanks!
I also searched for duplicates but didn't find any either...

BagelOrb, 2017-05-29:
Duplicate of https://github.com/Ultimaker/CuraEngine/issues/225

closed 2017-05-29 by BagelOrb