Ultimaker/Cura #1381: Cura 2.4.0 Mac (Ultimaten 2 Extended+) Nozzle Size

bulior, 2017-01-25:
Hey there, 

i try since days to get a variable Noozle Size within the Dropdown Menue ob the Um2 Extended plus Profile by edit the .json-files. Has someone done this before or is the solution to configure a custom FDM-Profile?

Best Regards

nallath, 2017-01-26:
What do you mean? There already is a dropdown menu with multiple nozzle sizes.

fieldOfView, 2017-01-26:
I think he's trying to make a custom nozzle size, eg 0.15 or 1.0.

nallath, 2017-01-26:
Ooh. Yeah, that takes a bit more work. You need to add a variant for this to work.

bulior, 2017-01-26:
Yeah I try it with the current Ultimaker variants but it would not be work.  Has some one done this before

Ghostkeeper, 2017-01-26:
Well, we created them so it was done before. I'd advise to start with copying `resources/variants/ultimaker2_extended_plus_0.25.inst.cfg` and copying all the 0.25mm profiles in `resources/quality/ultimaker2_plus`. Then modify the names, IDs and nozzle sizes of those files. See how far that gets you.

bulior, 2017-01-26:
Thanks for all comments!
I got it with the solution of Ghostkeeper. Thanks!

[http://imgur.com/a/3HqGj](`http://imgur.com/a/3HqGj`)

nallath, 2017-07-27:
This has now also been properly fixed.

closed 2017-07-27 by nallath