Ultimaker/Cura #1384: draft shield with more extruders

maukcc, 2017-01-26:
When draft shield is enabled, and you are printing whit multiple extruders the draft shield layers will be made with all extruders.
If you have loaded different materials in your extruders, these layers will not bond, and your draft shield layers will not stick and in worst case end up in your object.
The draft shield will have to be made with 1 extruder if multiple materials are loaded.

Ghostkeeper, 2017-01-26:
I agree. There's no real good reason why we're printing this with all extruders.

BagelOrb, 2017-01-27:
Now that I think about it again, there is some merit.

If the middle of your print only uses one extruder then you don't want to switch extruders every layer just for the draft shield.

Still it's best to keep using one extruder for layers which use both extruders anyway.

I'm including this in our internal issue tracker (CURA-3326).

Could you tell me what machine and materials you are using?

maukcc, 2017-01-27:
We have Cartesio machines, and use pretty much all materials. We do a lot of testing with PLA and PETG as support material, or visa versa.

BagelOrb, 2017-01-27:
So in which material do you want to print the draft shield? And which material you want to have removed from your draft shield?

maukcc, 2017-01-27:
One way of looking at this is that you choose the extruder, like you choose the extruder for support material.
Another way to look at it, is that the draft shield always uses the extruder that is going to be used 1st.
The 2nd option is best, but that does not mean it is always extruder_0

BagelOrb, 2017-01-27:
No I mean: what type of materials are currently in your draft shield and which one do you want removed from the draft shield?

What we could do is choose an extruder via a dropdown menu and then in a separate setting say whether it is the preferred extruder or required. This would then toggle between introducing extruder switches just to print the shield or settling with the current extruder at that layer.

BagelOrb, 2017-01-27:
For example: are you printing ABS-PLA and want PLA removed from the draft shield?

maukcc, 2017-01-27:
Which material is used is not relevant, Is it? 
For speed purposes it is relevant that it uses the extruder which is used 1st, as the other extruders may only be in use after a long while.

BagelOrb, 2017-01-27:
Well I don't know why you would want a draft shield and how bad the current situation is. If you are using some materials which Ultimaker doesn't support than it's not really high priority for us to fix this. I don't think our material testers are using draft shields. I wonder how much of a problem this is for our users.

I don't understand your comment. Do you mean 1st in the whole print, or 1st for every layer?

maukcc, 2017-01-27:
For us draft shield is used to bump ooze of the nozzle, and for keeping your object in a confined space.
For those purposes it is not relevant which material is being used for it. but it is relevant that it is being made with the extruder that starts the job at layer 0, as the other extruder(s) may not get involved until layer "X"
That means that it is not always extruder_0, as you can also have a raft that is being made with extruder_3, or other things that we can not foresee at this time

So probably best to have a drop down list of extruders, so the operator can choose : )

Ghostkeeper, 2017-01-27:
This would happen with any two materials that don't adhere well. Such as CPE-PVA, or PLA-ABS. These are supported by Ultimaker.

@BagelOrb in my original feature description (CURA-3325) I had this described, that we'd need an extruder setting (drop-down box) to select the extruder for the draft shield. You closed my issue as a duplicate and made your own issue (CURA-3326) without the proposed solution of an extruder selection drop-down.

BagelOrb, 2017-01-27:
There is no such thing as a build extruder. An extruder can be used to build the model with and at the same time be used to print support. If I have a dual color print, it's unclear which extruder would be classified as the build extruder.

Ghostkeeper, 2017-01-27:
We can probably figure something out like which object the draft shield is for but that gets complicated when multiple objects are close together. Hence I only mentioned the extruder selection setting in the feature description.

QuickOK, 2018-02-24:
Draft Shield - Draft Shield by extruder or material type... very simple. Will be looking for it in next beta release.

Legodev, 2021-04-19:
Since its almost 2 years, i would like to bump this, just to bring it back in discussion, since Cura with its implementation of a purge tower is still better than other slicers for multimaterial prints such as PLA+PETG or PLA+PVA. That the draft shield is printed with all tools, is the only issue.

Having a drop-down menu where one can select "Default" or "Extruder 1" or "Extruder 2" ... would be sufficient. (and also nice for ooze shields)

closed 2023-06-24 by github-actions[bot]