Ultimaker/Cura #1388: Loosing material setting, if PC idle for some time

structab, 2017-01-28:
If I leave the PC (win10) for a while (hour?) and the screen has gone black, then CURA has lost the connection to the printer (Ultimaker3) and reslice models with material 1. Ultimaker3 normal have PVA in PrintCore1, and that's normally not what I want to print with.

Ghostkeeper, 2017-01-30:
Oh, boy. This is going to be fun to test! 🤣

Ghostkeeper, 2017-01-31:
Turns out we already knew about this from an internal report. For reference of our developers: CURA-3059. It's scheduled for release 2.6 at the moment.

structab, 2017-01-31:
Thanks !
Any estimations om how long a "while" is ?

Ghostkeeper, 2017-01-31:
We try to release a release every 2 months, so it would be like for May or June. In practice it's a bit slower though.

Ghostkeeper, 2017-06-26:
This was just kicked off of our planning because it's been inactive for too long... I protested this because people hate having to re-authenticate, but it didn't make it nonetheless.

nallath, 2017-06-27:
The re authenticate issue should already be solved because of another issue report.

closed 2018-02-27 by ianpaschal