Ultimaker/Cura #1389: Duplicate model causes slicing error after copy is moved

bbum, 2017-01-29:
Starting with the STL from here:

http://www.thingiverse.com/thing:1863728

Drag the STL onto Cura.

Either drag a second time or duplicate.

Drag the second copy so it is aligned on the X axis with the first.

If you look closely, the X part of the vertical structure is no longer filled in on the copy.   The resulting print is unusable.

Cura 2.1.99-master.2016.09.30

[image]
[image]
[image]
[image]

bbum, 2017-01-29:
Realized I was running an ancient version.   Checked against current production release (2.3).   Bug still occurs.

fieldOfView, 2017-01-29:
I think this can be fixed by turning off one of the meshfixes that is turned on by default.

Ghostkeeper, 2017-01-31:
I don't really see the problem when I try this. It's got infill. It looks like it would print properly.

[image]

It's not really clear from the original screenshots either. Perhaps I'm misunderstanding what this is about. Is it about infill missing?

bbum, 2017-01-31:
@Ghostkeeper wrote:
> It's not really clear from the original screenshots either. Perhaps I'm misunderstanding what this is about. Is it about infill missing?

Look closely at the left hand tower in the last screenshot in my original post.

It isn't an infill issue, there are parts of the model flat out missing where the upright bits form the X.

Yours looks like it is correct.   What version of Cura are you using?   It may also be because of the mesh fix issue @fieldOfView mentioned.

Ghostkeeper, 2017-02-01:
I was using master, which is currently more or less the same as the 2.4 beta that you can find on the Ultimaker website. Perhaps it has been fixed since then. We did tweak some bits with wall outlines since 2.3.

bbum, 2017-02-01:
Thank you, @Ghostkeeper.

I'll give it a shot soon and will update this bug with the results.

closed 2017-07-27 by nallath