Ultimaker/Cura #1390: Cura save file name is different than shown

jmgurney, 2017-01-30:
Problem: I am using Fusion 360 w/ Cura to print.  When I hit save, it says it saved it w/ a file name different than the one displayed.

Version: 2.3.1 for MacOSX on 10.10.5

Expected behavior: The output file name be the same as it was displayed.

Steps to reproduce:
1.  Close Cura
2.  Using Fusion 360, select Make w/ 3D printer and output to Cura.
3.  Change the name in Cura to something more descriptive than the UUID that is used when exporting from Fusion 360.
4.  Save this and see that the file is named correctly.
5.  Using Fusion 360, select Make w/ 3D printer again, and output to Cura.
6.  Delete the original object, and recenter the new object.  (I have a model that the second one is off the bed, and so does not slice till it is recentered.)
7.  Observe that the name in the bottom right still shows the name set in step 3.
8.  Save the file, and notice that it saves using the UUID, not the one shown.

Ghostkeeper, 2017-01-31:
It doesn't seem to reproduce without Fusion 360. At least not with these reproduce steps:

1. Load Cube.stl. The job name in the bottom right is now "UMO_Cube" or something in that spirit.
2. Load Cylinder.stl. The job name is still UMO_Cube.
3. Delete the cube. The job name is still UMO_Cube.
4. Stick a USB stick in your computer.
5. In Cura, click Save to Removable Drive. The job name is still UMO_Cube.

The expected result is that it saves as UMO_Cube.gcode. This happens correctly.

I can't test with Fusion 360 since I don't have that application. Was it the plugin you wrote there, @thopiekar?

ianpaschal, 2018-02-26:
This is working in Cura 3.x

closed 2018-02-26 by ianpaschal