Ultimaker/Cura #1392: Request: Some way to identify which model is which after loading?

ntoff, 2017-01-30:
I tried hovering, right clicking, looking all about the UI but can't see anywhere that indicates which model is which or which model is currently selected. I have several which are very similar in appearance, some need duplicates others don't but once they're all imported it's hard to tell which one is which since they all look the same and there's no name list. Is my only option seriously to use another program to duplicate them, plate them, then import that into cura?

Using 2.4 beta on windows 10.

Ghostkeeper, 2017-01-31:
Eh, good one. Perhaps we could solve this by adding a non-clickable header to the context menu when you right-click a model or something?

There's a plug-in that does allow this among other things: https://github.com/sedwards2009/cura-god-mode-plugin

fieldOfView, 2017-02-02:
Or we could use the new legend area to show the names of loaded objects when in solid mode, indicating which models are selected by making them bold.

Ghostkeeper, 2017-02-02:
That would be more discoverable, but also takes some real estate though.

Ghostkeeper, 2017-06-05:
Our product manager just closed this for being inactive for 3 months.

nickthetait, 2017-06-07:
A little justification for this under-appreciated issue. :wink:  All from from my own personal experience.
* At least 8 different people have asked me for this feature (verbally, indirectly or otherwise)
* This feature would provide significant benefit to usability
* The level of effort is quite low. I suggest simply displaying a text label above the field for the project name

Ghostkeeper, 2017-06-10:
Relatedly, one of the designs for a future version of Cura featured a model tree, which shows all models on the build plate and would be clickable to select the model in question.

Ghostkeeper, 2017-07-10:
I think the model tree idea has been shot down by now. Haven't heard of it recently and they've been going in another direction with their designs.

alekseisasin, 2018-05-16:
Hi @ntoff , Cura has feature "Use multi build plate functionality". To enable it please open Preferences and you will find it in Experimental section. This feature allows to arrange models on the different build plates and identify models after loading.

ntoff, 2018-05-16:
Thanks, but unfortunately I moved away from Cura. Hopefully it'll help someone else though.

ianpaschal, 2018-05-25:
Closed in favor of discussion of #3535.

closed 2018-05-25 by ianpaschal