Ultimaker/Cura #1393: cura 2.3.1-2 segfault on startup in libQt5Gui

FirefighterBlu3, 2017-01-31:
Arch Linux, cura built from AUR after complete removal from system including user cache and config. dependencies also rebuilt.
```
[david@Scott ~]$ pb -Qi cura
Name            : cura
Version         : 2.3.1-2
Description     : A software solution for 3D printing aimed at RepRaps and the Ultimaker.
Architecture    : x86_64
URL             : https://ultimaker.com/en/products/cura-software
Licenses        : AGPLv3
Groups          : None
Provides        : cura
Depends On      : qt5-svg  python-scipy  python-pyserial  python-numpy  uranium  curaengine  cura-binary-data
Optional Deps   : None
Required By     : None
Optional For    : None
Conflicts With  : None
Replaces        : None
Installed Size  : 27.41 MiB
Packager        : Unknown Packager
Build Date      : Tue 31 Jan 2017 05:07:30 AM EST
Install Date    : Tue 31 Jan 2017 05:08:08 AM EST
Install Reason  : Explicitly installed
Install Script  : Yes
Validated By    : None
```

```
[david@Scott ~]$ pb -Qi python cura curaengine cura-binary-data uranium python-pyqt5 |awk '/Name|Version/ {print $0}'
Name            : python
Version         : 3.6.0-2
Name            : cura
Version         : 2.3.1-2
Name            : curaengine
Version         : 2.3.1-1
Name            : cura-binary-data
Version         : 2.3.1-2
Name            : uranium
Version         : 2.3.1-1
Name            : python-pyqt5
Version         : 5.7.1-3
```

```
[david@Scott ~]$ cura 
2017-01-31 05:19:26,512 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ConsoleLogger
2017-01-31 05:19:26,536 - INFO - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (__init__ [40]): Found CuraEngine at: /usr/bin/CuraEngine
2017-01-31 05:19:26,537 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraEngineBackend
2017-01-31 05:19:26,577 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin UpdateChecker
2017-01-31 05:19:26,577 - INFO - /usr/lib/uranium/plugins/UpdateChecker/UpdateChecker.py (run [28]): Checking for new version of cura
2017-01-31 05:19:26,584 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LocalFileOutputDevice
2017-01-31 05:19:26,584 - WARNING - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Plugin ConsoleLogger was already loaded
2017-01-31 05:19:26,587 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SimpleView
2017-01-31 05:19:26,589 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin WireframeView
2017-01-31 05:19:26,591 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin FileLogger
2017-01-31 05:19:26,596 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MirrorTool
2017-01-31 05:19:26,602 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin TranslateTool
2017-01-31 05:19:26,605 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SelectionTool
2017-01-31 05:19:26,608 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CameraTool
2017-01-31 05:19:26,616 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ScaleTool
2017-01-31 05:19:26,625 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RotateTool
2017-01-31 05:19:26,632 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFWriter
2017-01-31 05:19:26,635 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLWriter
2017-01-31 05:19:26,640 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLReader
2017-01-31 05:19:26,643 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJReader
2017-01-31 05:19:26,645 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJWriter
2017-01-31 05:19:26,660 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XmlMaterialProfile
2017-01-31 05:19:26,663 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LegacyProfileReader
2017-01-31 05:19:26,663 - WARNING - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Plugin UpdateChecker was already loaded
2017-01-31 05:19:26,663 - WARNING - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Plugin LocalFileOutputDevice was already loaded
2017-01-31 05:19:26,665 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileWriter
[... 158 lines cut ...]
   from /usr/lib/libQt5Qml.so.5
#14 0x00007f06b90f8137 in QQmlComponentPrivate::completeCreate() () from /usr/lib/libQt5Qml.so.5
#15 0x00007f06b90f7f7a in QQmlComponent::create(QQmlContext*) () from /usr/lib/libQt5Qml.so.5
#16 0x00007f06b916ec83 in QQmlApplicationEnginePrivate::_q_finishLoad(QObject*) () from /usr/lib/libQt5Qml.so.5
#17 0x00007f06b916ee92 in QQmlApplicationEnginePrivate::startLoad(QUrl const&, QByteArray const&, bool) () from /usr/lib/libQt5Qml.so.5
#18 0x00007f06b916eff0 in QQmlApplicationEngine::load(QString const&) () from /usr/lib/libQt5Qml.so.5
#19 0x00007f06b94f2660 in ?? () from /usr/lib/python3.6/site-packages/PyQt5/QtQml.so
#20 0x00007f06c867691f in _PyCFunction_FastCallDict () from /usr/lib/libpython3.6m.so.1.0
#21 0x00007f06c8685a0f in ?? () from /usr/lib/libpython3.6m.so.1.0
#22 0x00007f06c8641d77 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.6m.so.1.0
#23 0x00007f06c868553a in ?? () from /usr/lib/libpython3.6m.so.1.0
#24 0x00007f06c8685af3 in ?? () from /usr/lib/libpython3.6m.so.1.0
#25 0x00007f06c8641d77 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.6m.so.1.0
#26 0x00007f06c868553a in ?? () from /usr/lib/libpython3.6m.so.1.0
#27 0x00007f06c8685af3 in ?? () from /usr/lib/libpython3.6m.so.1.0
#28 0x00007f06c8641d77 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.6m.so.1.0
#29 0x00007f06c8685f47 in PyEval_EvalCodeEx () from /usr/lib/libpython3.6m.so.1.0
#30 0x00007f06c8641a5b in PyEval_EvalCode () from /usr/lib/libpython3.6m.so.1.0
#31 0x00007f06c870adc2 in ?? () from /usr/lib/libpython3.6m.so.1.0
#32 0x00007f06c870d62d in PyRun_FileExFlags () from /usr/lib/libpython3.6m.so.1.0
#33 0x00007f06c870d817 in PyRun_SimpleFileExFlags () from /usr/lib/libpython3.6m.so.1.0
#34 0x00007f06c87026f1 in Py_Main () from /usr/lib/libpython3.6m.so.1.0
#35 0x0000000000400a5d in main ()
(gdb) 
```

awhiemstra, 2017-01-31:
See also my comment here: https://github.com/Ultimaker/Cura/issues/1072#issuecomment-276347423

In short, this is due to a bug in PyQt which does not yet officially support Qt 5.8. There is currently no real known workaround.

ubitux, 2017-02-18:
This crash still happens with PyQt5.8 which is officially released since a few days: https://www.riverbankcomputing.com/software/pyqt/download5

greyltc, 2017-02-19:
Yeah. It seems like pyqt5.8 didn't fix it. I've rebuilt everything and I'm still getting a segfault:
```
python3[32422]: segfault at 10000001f ip 00007f964d634e39 sp 00007ffcb95059d0 error 4 in libQt5Gui.so.5.8.0[7f964d535000+522000]
systemd[1]: Started Process Core Dump (PID 32447/UID 0).
systemd-coredump[32448]: Process 32422 (python3) of user 1000 dumped core.
                                                
                                                Stack trace of thread 32422:
                                                #0  0x00007f964d634e39 _ZNK7QWindow8geometryEv (libQt5Gui.so.5)
                                                #1  0x00007f964d63a7c0 n/a (libQt5Gui.so.5)
                                                #2  0x00007f964f6ec25b n/a (libQt5Qml.so.5)
                                                #3  0x00007f964f6ecdd6 _ZNK3QV414QObjectWrapper14getQmlPropertyEP15QQmlContextDataPNS_6StringENS0_12RevisionModeEPbb (libQt5Qml.so.5)
                                                #4  0x00007f964f7028ff _ZN3QV47Runtime18method_getPropertyEPNS_15ExecutionEngineERKNS_5ValueEi (libQt5Qml.so.5)
                                                #5  0x00007f95ebebf717 n/a (n/a)
```

awhiemstra, 2017-02-20:
I have now reported the crash upstream, waiting to see what the reaction is.

greyltc, 2017-02-20:
@awhiemstra thanks. Can you give a link to your bug report?

awhiemstra, 2017-02-21:
There is no bugtracker for PyQt, everything goes through the mailing list. This is ML thread: https://www.riverbankcomputing.com/pipermail/pyqt/2017-February/038875.html

greyltc, 2017-02-21:
No wonder I couldn't find one!

ubitux, 2017-02-24:
So according to Phil Thompson (PyQt5 dev) on the ml, it was fixed in the latest snapshot (https://www.riverbankcomputing.com/pipermail/pyqt/2017-February/038909.html). So I tried that snapshot (`PyQt5_gpl-5.8.1.dev1702221543`) with includes the following change:

```--- a/qpy/QtQml/qpyqmlobject.cpp	2017-02-15 13:37:45.000000000 +0100
+++ b/qpy/QtQml/qpyqmlobject.cpp	2017-02-23 03:32:51.000000000 +0100
@@ -111,6 +111,13 @@
     // Note that we used to use sender() but this proved unreliable.
     if (call == QMetaObject::InvokeMetaMethod && proxied_mo->method(idx).methodType() == QMetaMethod::Signal)
     {
+        // Get the meta-object of the class that defines the signal.
+        while (idx < proxied_mo->methodOffset())
+        {
+            proxied_mo = proxied_mo->superClass();
+            Q_ASSERT(proxied_mo);
+        }
+
         // Relay the signal to QML.
         QMetaObject::activate(this, proxied_mo,
                 idx - proxied_mo->methodOffset(), args);
```

Unfortunately, it doesn't fix anything for me, not even the sample file provided. But maybe it's late and I missed something in the process... Can anyone else confirm?

ubitux, 2017-02-25:
Phil Thompson (PyQt5 dev) fixed the problem: https://www.riverbankcomputing.com/pipermail/pyqt/2017-February/038909.html

So I identified, extracted and tested the patch from the snapshot:

```diff
--- a/qpy/QtQml/qpyqml_register_type.cpp    2017-02-23 03:32:51.000000000 +0100
+++ b/qpy/QtQml/qpyqml_register_type.cpp    2017-02-25 03:32:58.000000000 +0100
@@ -124,13 +124,13 @@
 
 #define QPYQML_TYPE_INIT(n) \
     case n##U: \
-        QPyQmlObject##n::staticMetaObject = *mo; \
+        QPyQmlObject##n::staticMetaObject = static_mo; \
         QPyQmlObject##n::attachedPyType = attached; \
         rt->typeId = qRegisterNormalizedMetaType<QPyQmlObject##n *>(ptr_name); \
         rt->listId = qRegisterNormalizedMetaType<QQmlListProperty<QPyQmlObject##n> >(list_name); \
         rt->objectSize = ctor ? sizeof(QPyQmlObject##n) : 0; \
         if (ctor) rt->create = QQmlPrivate::createInto<QPyQmlObject##n>; else rt->create = 0; \
-        rt->metaObject = mo; \
+        rt->metaObject = &QPyQmlObject##n::staticMetaObject; \
         rt->attachedPropertiesFunction = attached_mo ? QPyQmlObject##n::attachedProperties : 0; \
         rt->attachedPropertiesMetaObject = attached_mo; \
         rt->parserStatusCast = is_parser_status ? QQmlPrivate::StaticCastSelector<QPyQmlObject##n,QQmlParserStatus>::cast() : -1; \
@@ -219,7 +219,33 @@
         return 0;
     }
 
-    const QMetaObject *mo = pyqt5_qtqml_get_qmetaobject(py_type);
+    const QMetaObject *orig_mo = pyqt5_qtqml_get_qmetaobject(py_type);
+    QMetaObject static_mo = *orig_mo;
+
+#if QT_VERSION >= 0x050800
+    // Qt v5.8.0 changed the way properties are handled by directly calling a
+    // class's static meta-call (if there was one) directly.  This bypasses the
+    // proxy and calls the static meta-call with a pointer to the proxy rather
+    // than a pointer to the real object.  To work round this we clone the
+    // QMetaObject chain and remove the references to the static meta-call
+    // forcing the earlier behaviour.  This approach may also work with earlier
+    // versions of Qt - but if it ain't broke...
+    static_mo.d.static_metacall = 0;
+
+    QMetaObject *sub_mo = &static_mo;
+
+    // By retaining the QObject static meta-object we don't appear to be a
+    // gadget.
+    for (const QMetaObject *mo = sub_mo->d.superdata; mo != &QObject::staticMetaObject; mo = mo->d.superdata)
+    {
+        QMetaObject *new_mo = new QMetaObject;
+        *new_mo = *mo;
+        new_mo->d.static_metacall = 0;
+
+        sub_mo->d.superdata = new_mo;
+        sub_mo = new_mo;
+    }
+#endif
 
     // See if the type is a parser status.
     bool is_parser_status = PyType_IsSubtype(py_type,
@@ -272,7 +298,7 @@
 
     if (qquickitem_register)
     {
-        sipErrorState estate = qquickitem_register(py_type, mo, ptr_name,
+        sipErrorState estate = qquickitem_register(py_type, orig_mo, ptr_name,
                 list_name, &rt);
 
         if (estate == sipErrorFail)

```

The issue is fixed for me in Cura with this. I sent a mail to the Archlinux packager so this patch can be included.

Thanks

greyltc, 2017-02-25:
Cool! I not so keen on waiting, so I put a pyqt5-hotfix package with that patch into the AUR and made the cura package depend on it. 2.4.0 launches without segfault now.

I think this issue can be closed.

Ghostkeeper, 2017-03-01:
We'll upgrade to PyQt 5.8 when we're ready. Until then, Arch users can compile with 5.8 themselves or use our builds with 5.7.

closed 2017-03-01 by Ghostkeeper